bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #1: Firmware

LemaruX, 2019-07-01:
Please release the uncompiled firmware so we can edit settings and compile for ourselves. Thanks.

bjoernwedi, 2019-07-02:
Need to change the steppers too

Promite2501, 2019-07-05:
I think this is meant to be more for their how to PDFs and basic info.  If you want to compile from scratch its is highly recommended to start with the main marlin branch of 2.0. 
[https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x](url)

LemaruX, 2019-07-05:
Yeah, I'd always use latest vanilla marlin anyway. Just past experiences with my other printers have proven that often they set custom values in their own version of marlin which would be nice to copy over to the original marlin source.

claudiusa, 2019-07-16:
Bigtreetech released the uncompiled Marlin firmware with board pins configuration today.

thisiskeithb, 2019-07-16:
It should be in the main Marlin repo soon: https://github.com/MarlinFirmware/Marlin/pull/14638

TheZoker, 2019-07-31:
@LemaruX I guess you can close this issue, since everything is in the repo

oldman4U, 2020-06-08:
@thinkyhead (Please close this ticket due to inactivity)

Thank you

1st try;-)

closed 2020-06-08 by LemaruX