bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #2: Bootloader

Bansh1990, 2019-07-28:
HI, bootloader has been erased. i cant flash with sd card, and cant connect over USB.
Can you share bootloader.bin with guide for flashing to recover my board?

Bansh1990, 2019-07-29:
yes? of course.
[SKR-E3-SIP-bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3440956/SKR-E3-SIP-bootloader.zip)
 there is.

flash from The address is 0x08000000

Bansh1990, 2019-07-29:
trhank you

SSPW, 2019-09-26:
> yes, of course.
> [SKR-E3-SIP-bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3440956/SKR-E3-SIP-bootloader.zip)
> there is.
> 
> flash from The address is 0x08000000

is this bootloader for the RC or RE.
apparently E3 DIP is shipping with a 512K flash RE now, like i received

When i use this bootloader, the firmware is flashed.
But the firmware.bin file is not removed, so it reflashes every boot.
The FIRMWARE.CUR behaviour is unstable. Sometimes its missing.

lococnc, 2019-09-30:
I found that with as many times as I have flashed the firmware that about 1 in 10 fails like described with endless beep and blank screen.
I formatted the SD card in my PC, then copied the files to it fresh and it will flash successfully. However it does occasionally not rename to .cur

thubot, 2019-10-02:
Here is the bootloader that came with my board that I received today, which shipped with an STM32F103RET6. In a hexdiff I noticed that it seems to be a different version of the one above but didn't look further into that.

[SKR-E3-DIP-firmware-and-bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3679015/SKR-E3-DIP-firmware-and-bootloader.zip)

Piscanc, 2019-11-01:
Hi, anyone up for a challenge? ;D
 
i have similar problem... bricked board (rct6 variant).
board behavior: 
status leds always on (both red and blue),not detected by PC, Ender3 screen only lit and buzzing, btt touchscreen not detecting the board (if reset pin inserted, screen goes in reset), sd slot not responding.
I can connect to it over st-link, but that is all i can do... it recognizes stm, reads the blocks, but they are empty. 
from St-link Utility i get this 
20:36:58 : Core is held in reset
20:36:58 : Error occured during program operation!
20:37:14 : Programmed memory Checksum: 0x00000000

from CubeProgrammer i first get flash memory protection, (after disabling it) i get failed to erase memory and lastly if i skip that... it hangs during writing in first block, i have to disconnect.
full log from connecting:
21:03:04:205 : STLinkUSBDriver.dll loaded
21:03:04:210 : STLinkUSBDriver.dll loaded
21:03:04:211 : ST-LINK SN : 51FF6C067871515223181867
21:03:04:212 : ST-LINK FW : V2J34S7
21:03:04:212 : Voltage : 3.22V
21:03:04:218 : SWD freq : 4000 KHz
21:03:04:219 : Connect mode: Hot Plug
21:03:04:222 : Reset mode : Hardware reset
21:03:04:553 : Device ID : 0x414
21:03:05:202 : flash loader C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeProgrammer\bin/FlashLoader/0x414.stldr is loaded
21:03:05:685 : UPLOADING OPTION BYTES DATA ...
21:03:05:686 : Bank : 0x00
21:03:05:687 : Address : 0x4002201c
21:03:05:688 : Size : 8 Bytes
21:03:05:689 : Reading data...
21:03:05:689 : r ap 0 @0x4002201C 0x00000008 bytes
21:03:05:691 : Bank : 0x01
21:03:05:691 : Address : 0x1ffff800
21:03:05:692 : Size : 16 Bytes
21:03:05:693 : Reading data...
21:03:05:694 : r ap 0 @0x1FFFF800 0x00000010 bytes
21:03:05:696 : UPLOADING OPTION BYTES DATA ...
21:03:05:697 : Bank : 0x00
21:03:05:698 : Address : 0x4002201c
21:03:05:699 : Size : 8 Bytes
21:03:05:700 : Reading data...
21:03:05:701 : r ap 0 @0x4002201C 0x00000008 bytes
21:03:05:703 : Bank : 0x01
21:03:05:703 : Address : 0x1ffff800
21:03:05:705 : Size : 16 Bytes
21:03:05:706 : Reading data...
21:03:05:707 : r ap 0 @0x1FFFF800 0x00000010 bytes
21:03:05:715 : UPLOADING ...
21:03:05:716 : Size : 1024 Bytes
21:03:05:716 : Address : 0x8000000
21:03:05:717 : Read progress:
21:03:05:718 : Reading data...
21:03:05:718 : r ap 0 @0x08000000 0x00000400 bytes
21:03:05:719 : Data read successfully
21:03:05:720 : Time elapsed during the read operation is: 00:00:00.012

full log from write attempt without skipping erase:
21:04:27:297 : Memory Programming ...
21:04:27:298 : Opening and parsing file: bootloader-bigtreetech-e3-dip.bin
21:04:27:299 : File : bootloader-bigtreetech-e3-dip.bin
21:04:27:300 : Size : 17096 Bytes
21:04:27:300 : Address : 0x08000000
21:04:27:302 : Erasing Segment <0> Address <0x08000000> Size <17096>Bytes
21:04:27:302 : Erasing memory corresponding to segment 0:
21:04:27:688 : Memory erase...
21:04:30:001 : halt ap 0
21:04:30:002 : w ap 0 reg 15 PC (0x20000000)
21:04:30:003 : w ap 0 reg 17 MSP (0x20000500)
21:04:30:006 : w ap 0 reg 16 xPSR (0x01000000)
21:04:30:023 : w ap 0 @0x20000960 0x00000200 bytes
21:04:30:030 : w ap 0 @0x20000000 0x00000004 bytes
21:04:30:033 : w ap 0 @0x20000004 0x0000053E bytes
21:04:30:034 : Erasing internal memory sectors [0 8]
21:04:30:035 : Init flashloader...
21:04:32:001 : halt ap 0
21:04:32:002 : halt ap 0
21:04:34:001 : w ap 0 reg 0 R0 0x00000001
21:04:34:002 : w ap 0 reg 1 R1 0x00000000
21:04:34:010 : w ap 0 reg 2 R2 0x00000000
21:04:34:017 : w ap 0 reg 3 R3 0x00000000
21:04:34:017 : w ap 0 reg 4 R4 0x00000000
21:04:34:017 : w ap 0 reg 5 R5 0x00000000
21:04:34:018 : w ap 0 reg 6 R6 0x00000000
21:04:34:018 : w ap 0 reg 7 R7 0x00000000
21:04:34:019 : w ap 0 reg 8 R8 0x00000000
21:04:34:019 : w ap 0 reg 9 R9 0x00000000
21:04:34:020 : w ap 0 reg 10 R10 0x00000000
21:04:34:020 : w ap 0 reg 11 R11 0x00000000
21:04:34:026 : w ap 0 reg 12 R12 0x00000000
21:04:34:027 : w ap 0 reg 13 SP 0x00000000
21:04:34:028 : w ap 0 reg 14 LR 0x20000001
21:04:34:028 : w ap 0 reg 15 PC 0x20000005
21:04:34:029 : w ap 0 reg 16 xPSR 0x01000000
21:04:34:029 : w ap 0 reg 17 MSP 0x2000093E
21:04:34:030 : w ap 0 reg 18 PSP 0x00000000
21:04:34:031 : run ap 0
21:04:36:000 : halt ap 0
21:04:36:001 : r ap 0 reg 0 R0 0x00000000
21:04:36:002 : Init function fail
21:04:36:012 : r ap 0 reg 16 xPSR 0x00000000
21:04:36:013 : Loader sector erase...
21:04:36:013 : Error: failed to erase memory
21:04:36:082 : Error: failed to erase memory

from what i can tell boot0 is in low state, but i dont see any way in changing that.
Is there any help for me, besides swaping parts?

ps: sorry for lengthy post

TheAssassin, 2019-11-04:
For some reason, noone's posted their flash commands yet. I hope this will help people in the same situation.

I've struggled for 2 days with my board, which came without firmware and bootloader. Flashing didn't work, communication wasn't possible (USB wasn't even recognized), so I suspected a bootloader problem very early. I then hooked up some debugger (a black magic probe) and could run the software (via `pio debug`, just had to set a few things in the `platformio.ini`), however nothing else worked, and I was left with a board that I could only run by starting the software from another computer. A super annoying situation, especially since I had a brand new board.

I sourced an ST-Link v2 from a friend, and started working on the issue. After hooking it up to the SWD port on the board (luckily there's such a port, there's no other means for debugging, such as a UART interface for serial comunication) with four jumper cables (GND, 3.3V, SWDIO and SWCLK, make sure it's correct or the board won't be found!), I was able to flash the bootloader + firmware from the zip archive provided above.

For flashing on Linux, I can recommend using [stlink](https://github.com/texane/stlink), a free/open source tool that can flash STM32 chips over SWD. I struggled a bit finding the start address, which is very important if you have `.bin` files and not `.hex` (the latter provides such information already). With the wrong start address, `st-flash` just prints `Unknown memory region`, which doesn't really help.

The following command worked, flashing went pretty uneventful:

```sh
> st-flash write firmware-bigtreetech-e3-dip-stock-with-bootloader.bin 0x8000000
st-flash 1.5.1-47-g2901826
2019-11-04T15:21:45 INFO common.c: Loading device parameters....
2019-11-04T15:21:45 INFO common.c: Device connected is: F1 High-density device, id 0x10036414
2019-11-04T15:21:45 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 2048 bytes
2019-11-04T15:21:45 INFO common.c: Ignoring 279996 bytes of 0xff at end of file
2019-11-04T15:21:45 INFO common.c: Attempting to write 244292 (0x3ba44) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0803b800 erased
2019-11-04T15:21:47 INFO common.c: Finished erasing 120 pages of 2048 (0x800) bytes
2019-11-04T15:21:47 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2019-11-04T15:21:47 INFO flash_loader.c: Successfully loaded flash loader in sram
120/120 pages written
2019-11-04T15:21:57 INFO common.c: Starting verification of write complete
2019-11-04T15:21:59 INFO common.c: Flash written and verified! jolly good!
```

The start address `0x8000000` is where the bootloader starts. It occupies `0x4000` bytes (16kiB), so naturally, the actual firmware starts at `0x8004000`. Since we need the bootloader as well, we need to flash it there. You can just flash the bootloader if you plan to use your own firmware, too, it's sufficient.

It's disappointing @bigtreetech didn't chime in in this discussion, providing the missing bootloader + firmware and instructions. @bigtreetech your boards are praised on the Internet, I hope your quality assurance is going to be improved. Boards without a firmware are one thing, but boards without a bootloader are *really* annoying. You need to invest time and money in order to fix them, and that shouldn't be required.

P.S.: Huge thanks to @thubot for providing the original firmware. I suspect you read it out with an ST-Link v2 programmer as well. After finding the right command, I could flash it just fine with an ST-Link v2 and [stlink](https://github.com/texane/stlink). The bootloader then worked as intended, flashing a custom Marlin build worked fine from the SD card, the blue LED was flashing just fine, the file was renamed and after fixing my Marlin config, I could communicate with the board via USB and a serial console (using the excellent [tio](https://tio.github.io/), which I can really recommend for serial communication, it's a life saver!) as expected. I'm going to plug everything in again and will try it out asap!

P.P.S.: @Piscanc try my instructions on a Linux computer (a live CD will work just fine), I hope you can get your board flashed as well.

P.P.P.S.: If you don't have an ST-Link v2 and don't want to invest 2-3 bucks for a cheap Chinese one, please look for the next FabLab or Hackerspace, they most likely have one and they can probably help you flashing it!

Piscanc, 2019-11-04:
@TheAssassin thank you for your notice. I must confess, i´m not "sw" person... i would rather cut peace of something few times, than making 3d model... if you know what i mean, but i am trying... 
I managed to get st-link 1.5.1 on ubuntu live 19.04, but i get same  Unknown memory region    :(
getting your git link version running is beyond me, al least without more detailed guidance...

to get it in the open...
i got 3.5tft v2.0 (i presume that those who have it, know what i mean) and this board on the local "flee" market(eu)... but at last this board was sold as non-communicating... and was warned, but i rolled the dice. Thinking if i managed to get tft working (presumably, no way to test it atm), loading up this board would be walk in the park. It has everything clearly labeled, it read it right up, just...

to sum up, i have st-linkv2 directly from farrnell, so it should be god,
stm32 chip is recognized/connected in windows and in linux,
basic st-utility stops at core is held in reset, or freezes if i skip erasing or verify
bit more "comprehensive" Cube programmer does the same, first it complains that its read protected, then it fails to erase memory, if i disable protection there it freezes again. i would understand if something was written in the blocks, but all are 0. 
would this say bad chip or empty chip? am i missing something obvious? you cant write, if you cant erase. 

i am not going to trouble of obtaining it and swapping chips...  

PS: tnx to every one who participates :)

TheAssassin, 2019-11-04:
Please post your command and your log of st-link on Ubuntu, @Piscanc. Probably just a usage error. We must get your board to work. Mine's printing now just fine in my Ender-2 (despite the display isn't working, might need to get an ender-3 one).

Piscanc, 2019-11-05:
@TheAssassin 
ubuntu@ubuntu:~/Downloads$ st-info --probe
Found 1 stlink programmers
 serial: 513f6c067871515223181867
openocd: "\x51\x3f\x6c\x06\x78\x71\x51\x52\x23\x18\x18\x67"
  flash: 0 (pagesize: 2048)
   sram: 65536
 chipid: 0x0414
  descr: F1 High-density device
ubuntu@ubuntu:~/Downloads$ st-flash write firmware-bigtreetech-e3-dip-stock-with-bootloader.bin 0x8000000
st-flash 1.5.1
2019-11-05T18:51:50 INFO common.c: Loading device parameters....
2019-11-05T18:51:50 INFO common.c: Device connected is: F1 High-density device, id 0x10036414
2019-11-05T18:51:50 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0 bytes (0 KiB) in pages of 2048 bytes
Unknown memory region
ubuntu@ubuntu:~/Downloads$ sudo st-flash write firmware-bigtreetech-e3-dip-stock-with-bootloader.bin 0x8000000st-flash 1.5.1
2019-11-05T18:52:02 INFO common.c: Loading device parameters....
2019-11-05T18:52:03 INFO common.c: Device connected is: F1 High-density device, id 0x10036414
2019-11-05T18:52:03 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0 bytes (0 KiB) in pages of 2048 bytes
Unknown memory region

hope it helps...

TheAssassin, 2019-11-05:
> flash: 0 (pagesize: 2048)

> 2019-11-05T18:52:03 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0 bytes (0 KiB) in

That doesn't look good indeed. The value of "flash" should be 256KiB or 512KiB at least. It might indeed be a faulty chip.

Can you read what's printed on the chip?

Piscanc, 2019-11-05:
STM32F103
RCT6
99222   98
MYS    99   837

TheAssassin, 2019-11-05:
You can try one last time with the one in the zip provided [here](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/2#issuecomment-515893243). But to be honest, this looks like your chip is faulty. 

You can, however, try to ask in the stlink issue tracker about the flash being 0 bytes. Perhaps it's just a software bug.

Piscanc, 2019-11-05:
Tried that... even tried wrog one ment for 2209, just to see. But same problem...

Now was it because of shipping, esd, or sun flare... or error during first initial firmware load? Witch was done with SD provided.

TheAssassin, 2019-11-05:
> Now was it because of shipping, esd, or sun flare... or error during first initial firmware load? Witch was done with SD provided.

I doubt it's your fault. I also doubt you ever succeeded in flashing firmware. I mean, your board likely has been lacking a bootloader to begin with. The bootloader is the piece of software that is loaded first, and that flashes the actual firmware from the SD card. When it's missing, the board is pretty much "dead".

Piscanc, 2019-11-05:
:) agree, that i didnt mess up the board, because it was too broken to begin with.
but i could make an argument that every chip is "born" empty, we just lack skill/equipment to do so...

TheAssassin, 2019-11-05:
The `flash` parameter is the size of the space you can flash. If it displays `0`, it means you can't write to it. That's clearly an issue with the hardware. Check my log, it tells the flash size is 512 kiB (I got a newer chip which has a larger flash area, yours should show 256 kiB as per the specification).

sl1pkn07, 2019-11-15:
thi @TheAssassin . the RET6 bootloader can't flash the firmware into the E3DIP (RET6 unit) if is more greather than 256KB (like the "official" mcu capacity have), rigth?

any tip (with commands if is possible (and linux xd)) for do it?

my firmware is more tan 256K and if put it into the SD card and reboot, do nothing (uppercase/lowercase test)

greetings

TheAssassin, 2019-11-15:
> thi @TheAssassin . the RET6 bootloader can't flash the firmware into the E3DIP (RET6 unit) if is more greather than 256KB (like the "official" mcu capacity have), rigth?

I've just flashed the bootloader provided in that comment. The bootloader itself is just 16 kiB, what you talk about is the ROM size including the firmware (but even that is smaller than 256 kiB).

It could be that the bootloader doesn't want to flash your firmware, but flashing is definitely possible over the SWD pins. Unfortunately, we can't change the bootloader, we didn't even get an official binary, someone had to read out theirs from the board. But I'd try to trim down the firmware. Otherwise you need an ST-Link v2 (clone) (a bluepill board might work as well, there's guides on the Internet how to use them as ST-Link v2).

After having gone through all of this, I already thought about trying to write my own bootloader that flashes from the SD card. There's examples on the Internet how to do so.

By the way, I believe the reason that their upstream bootloader refuses to flash firmwares with > 256 kiB even when you have a RET6 is that they just use the same one for all boards. I assume there were no more RCT6s and they simply took the next size to continue manufacturing. When you have more than one chip with different flash sizes, you have to write your bootloader so that it works on all chips, right?

sl1pkn07, 2019-11-15:
is possible use this one https://github.com/DAlexis/stm32-usb-bootloader instead the official (in RET6 units)? seems have the same options like the official one (update from USB (treated as storage mass when conect to pc) and from SD card. and is native RET6 mcu

the unique thing different, is the name of the formware, but is possible change it

greetings

TheAssassin, 2019-11-15:
I haven't tried that. In theory any STM32 bootloader should work.

sl1pkn07, 2019-11-15:
is strange. just build (and do some changes about spaces in the namepaths i the build script and cmakefile), with gcc-none-eabi 9.2.0 (ship by my distro) and get a 510KB file? is normal?

TheAssassin, 2019-11-16:
Don't know *what* you're building there. I'm using https://github.com/TheAssassin/Marlin-Ender2-SKR-E3-DIP, and end up with a file size of 210 kiB or so...

sl1pkn07, 2019-11-16:
I talk about that bootloader

TheAssassin, 2019-11-16:
Strange, when I build this using that script, I end up with much smaller files:

```sh
> du -sh build/*/*.bin
48K	build/debug/usb-bootloader.bin
48K	build/release/usb-bootloader.bin
```

sl1pkn07, 2019-11-16:
[image]

mmmm

TheAssassin, 2019-11-16:
Use the script.

In any case, this is all way off topic. If you have further questions, please consider reaching out to me via IRC on FreeNode.

sl1pkn07, 2019-11-16:
my screenshot is taked from CMakeCache.txt loaded from the build.sh script

sl1pkn07, 2019-11-19:
> Aquí está el gestor de arranque que vino con mi placa que recibí hoy, que se envió con un STM32F103RET6. En un hexdiff noté que parece ser una versión diferente de la anterior, pero no busqué más en eso.
> 
> [SKR-E3-DIP-firmware-and-bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3679015/SKR-E3-DIP-firmware-and-bootloader.zip)

seems the bootloader (only) is broken. i've flash it and get brick (restore using openocd and push the reset button durning the procedure)

TheAssassin, 2019-11-19:
Yeah, I also mentioned I flashed the bootloader and firmware bin. Thanks for testing the bootloader only.

sl1pkn07, 2020-04-05:
https://github.com/sl1pkn07/SKR_E3_DIP_Bootloader

please test it (ONLY for SKE E3 DIP 1.0 with RET6 MCU)

needs build with stm32cubeide

olegneo, 2020-04-07:
Hi2all!
Please give me a link where to download the bootloader to the SKR E3-DIP v1.1 board

TheAssassin, 2020-04-08:
@olegneo check what microcontroller you have on the board. The bootloader @thubot provided works well enough, you could try that. Please beware you need an ST-Link v2 (or cheap clone) to flash the board.

olegneo, 2020-04-08:
> @olegneo check what microcontroller you have on the board. The bootloader @thubot provided works well enough, you could try that. Please beware you need an ST-Link v2 (or cheap clone) to flash the board.
STM32F103RET6

Piscanc, 2020-04-09:
since i got e3_mini, i kinda lost interest in DIP...
i could test again in a few days since i have ST-link at my work, i am stuck at home...
perhaps it could be done with USBTiny? but that would mean another big variable that i really dont have for. 
printing visors for local maker comunity, assembling another homemade printer, work round the house...

stay well everyone

JuStAdsor, 2020-04-11:
Hi!

I have just ordered an e3 mini, but I want to change the MCU to the one with 512k.
Now, what bootloader should I flash so I can take advantage of the all 512kb?

Thank you!

TheAssassin, 2020-04-11:
Seems like a weird plan, 256kiB is plenty space for Marlin.

Anyway, @sl1pkn07 wrote one for the RET6, which is the one with 512kiB of flash. If you solder in one of these, you could try his. I have a RET6 board, too, but I'm using the original bootloader (which is potentially for the RCT6) without any issues.

sl1pkn07, 2020-04-11:
the mine is experimental. as far i check, only works the first time, the second time only boot the flashed firm in first instance.

i'm not coder, absolutely not. is only a modified version from guy in ST forums

if any with coder skill can catch a eye to te code, is will help a lot

greetings

JuStAdsor, 2020-04-11:
> the mine is experimental. as far i check, only works the first time, the second time only boot the flashed firm in first instance.
> 
> i'm not coder, absolutely not. is only a modified version from guy in ST forums
> 
> if any with coder skill can catch a eye into the code, will help a lot
> 
> greetings

It would be great. But where can we find the source code for the bootloader?

sl1pkn07, 2020-04-11:
the original (comes with BTT board) no exist, the mine is here https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/2#issuecomment-609368756

JuStAdsor, 2020-04-11:
> the original (comes with BTT board) no exist, the mine is here [#2 (comment)](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/2#issuecomment-609368756)

I believe you need to adjust this line:
FLASH (rx)      : ORIGIN = 0x08000000, LENGTH = 512K
To this
FLASH (rx)      : ORIGIN = 0x08006000, LENGTH = 512K - 0x6000

So this way you don't overight the bootloader when you flash the firmware.

sl1pkn07, 2020-04-11:
in here?

https://github.com/MarlinFirmware/Marlin/blob/0518dec60d0931745efa2812fa388f33d68cfa29/Marlin/src/pins/stm32f1/pins_BTT_SKR_E3_DIP.h#L37-L38

ive already test dump the bootloader (after flashing the firmware throught bootloader(SD)) with stm32cubeprogrammer from 0x08000000 to 0x08007000 and the dump is the same except the latest 2 or 3 bits

[image]

~~~
┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[~/aplicaciones/Creality Ender 3/SKR_E3_DIP_Bootloader/SDBootloader/Release]|
└───╼  hexdump SDBootloader.bin > test.hex
┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[~/aplicaciones/Creality Ender 3/SKR_E3_DIP_Bootloader/SDBootloader/Release]|
└───╼  hexdump test > test2.hex
┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[~/aplicaciones/Creality Ender 3/SKR_E3_DIP_Bootloader/SDBootloader/Release]|
└───╼  kompare test*hex
~~~

[image]

then the bootloader is not overwrite when flash the firmware througth bootloader

JuStAdsor, 2020-04-11:
> in here?
> 
> https://github.com/MarlinFirmware/Marlin/blob/0518dec60d0931745efa2812fa388f33d68cfa29/Marlin/src/pins/stm32f1/pins_BTT_SKR_E3_DIP.h#L37-L38
> 
> ive already test dump the bootloader (after flashing the firmware throught bootloader(SD)) with stm32cubeprogrammer from 0x08000000 to 0x08007000 and the dump is the same except the latest 2 or 3 bits
> 
> [image]
> 
> ```
> ┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[~/aplicaciones/Creality Ender 3/SKR_E3_DIP_Bootloader/SDBootloader/Release]|
> └───╼  hexdump SDBootloader.bin > test.hex
> ┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[~/aplicaciones/Creality Ender 3/SKR_E3_DIP_Bootloader/SDBootloader/Release]|
> └───╼  hexdump test > test2.hex
> ┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[~/aplicaciones/Creality Ender 3/SKR_E3_DIP_Bootloader/SDBootloader/Release]|
> └───╼  kompare test*hex
> ```
> 
> [image]
> 
> then the bootloader is not overwrite when flash the firmware througth bootloader
> 
> edit. i have also this modification in the marlin
> 
> [image]

Than is really strange why the first time works but after it doesn't :(

sl1pkn07, 2020-04-11:
yep. but working again if erase flash from 0x08007000 to the end (with stm32cubeprogrammer) (make sure kepping 0x08000000 to 0x08006FF0 untounch). but fail again when the formware is flashed

wisesokol, 2020-05-27:
Has anyone solved the problem with bootloader? I want to replace RCT6 to RET6 on skr mini e3 v1.2

Piscanc, 2020-05-29:
been trying to load mine again... no succes.
my space size is listed as 0 or unknown?!?
cubeprogrammer finds separate blocks but they are protected. If i try to rewrite it, it just freezes and i have to disconnect st-link to wake it up again.
different programming tools reacts similar, "freezing" or trowing write error... 
after additional digging i found possible solution "j-link" 
https://wiki.segger.com/STM32#Disabling_readout_protection
supposedly their sw can override protection.
there was also a post witch i lost :( ...on one automotive forum with instructions for converting st-link-v2/clone to j-link with register hack. i didnt save it because i original st-link from work, witch i "cant" mod. 
so i am out of this game with this board, far too much time has gone into debugging, it will sit in a drawer as spare parts if something goes wrong with e3_mini_v1.2

ss89, 2020-06-28:
> Here is the bootloader that came with my board that I received today, which shipped with an STM32F103RET6. In a hexdiff I noticed that it seems to be a different version of the one above but didn't [...]

Thank you so much for your bootloader. My e3 dip didn't want to take bigger firmware files, but now it does :)

wisesokol, 2020-06-28:
You are mistaken.  The existing bootloader is written to a maximum of 256 kb.  For now, you can only write large firmware via stlink. Most likely, the firmware was not recorded completely. There may be problems when working.

wisesokol, 2020-07-12:
New info for me.
[https://www.youtube.com/watch?v=q0JEx3uzgSo](url)
Bootloder can flash more than 256.

wisesokol, 2020-07-19:
Skr mini E3 v1.2 bootloader

[bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4943846/bootloader.zip)

I change RCT to RET chip on my board. This bootloader supports firmware over 255 kB. 
I could not do more, since I have already included everything that I can in the firmware.
[image]

penright, 2020-07-21:
I think I too have lost my boot loader. I have ST-Link V2.  I may have missed it, but most of these comments assume STLink experience. The wire seems straight forward, 3.3-3.3, gnd-gnd, swclk-clk, and swdio-dio. The thread talks about Linux base flashing software, is there a windows version? I can load a Linux VM, it would just save me a few steps. So I am currently setting up PIO. I download the zip file and there are 3 files, the readme, bootloader, stock with bootloader. Assuming the names and looking at the sizes, all I need is to flash "bootloader-bigtreetech-e3-dip.bin" correct? Any hints on using VSCode/PIO ?

sl1pkn07, 2020-07-21:
https://www.st.com/en/development-tools/stm32cubeprog.html
or
https://www.st.com/en/development-tools/stsw-link004.html

wisesokol, 2020-07-22:
I will release a detailed video very soon. Wait a couple of days

penright, 2020-07-22:
> I will release a detailed video very soon. Wait a couple of days

Cool. I have been playing with the "flashing", now and I thought I had a bad STLink-V2. When I plug it in, windows does not understand the device. I have installed the ST drivers. I found where you can flash a Blue Pill. I can connect to it serially and flash it whit the hex file provided which may or may not be right. But when I try and connect to use it USB, same thing. I am ordering one more clone and a Nucleo. Thanks for documenting this process.

penright, 2020-07-22:
> I will release a detailed video very soon. Wait a couple of days
If you tell me to hold my horses, I understand ...
Excited about the video, but meanwhile, I have been playing around. I think I am starting to understand. Looking at the schematics for the E3 DIP. The Boot0 is hardwired low.  Am I right in saying to use either the serial (USB) or SW (STLink) that has to be high? How do you do that? Looks like the Boot1 (P2) is the Z DIR, if yes, reading print straight up, reader header, right before the EN for the extruder. Am I getting close?
I did get my Blue Pill flashed, then updated, and now windows see it when plugged in and so does the cube programmer. Progress.

wisesokol, 2020-07-23:
Hi All.
Video is done.
[https://www.reddit.com/r/ender3/comments/hwi0w0/skr_mini_e3_upgrade_to_ret6_512kb/](url)

abtx, 2020-08-01:
Any tutorial on how to flash the board? I've got BTT SKR Mini E3 V1.2 and it seems bricked.

wisesokol, 2020-08-01:
> Any tutorial on how to flash the board? I've got BTT SKR Mini E3 V1.2 and it seems bricked.

See my video. Link in my post upper

abtx, 2020-08-02:
@wisesokol thanks but the link doesn't seem to link to anything. Can you check it please?

wisesokol, 2020-08-02:
> @wisesokol thanks but the link doesn't seem to link to anything. Can you check it please?
Shit linker!!!!

https://youtu.be/w2hJrD5voBY
Copy and paste in to browser

[18 more replies]