bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #4: SKR Mini E3 DIP USB connection on startup

pyunramura, 2019-08-02:
I'm having a somewhat similar issue described by @Xenon-2510 in https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/issues/2#issuecomment-513290031_ .

Upon startup of the printer with USB host attached and connected with either a Windows desktop, OSX laptop, or Raspberry Pi with Octopi I'm seeing a non-sane value for the hotend temperature (-14c in my case) on both host and printer which trigger the buzzer speaker continuously upon changing the hotend temperature via pronterface, octopi, or the printer itself. I'm not certain if it has to do with thermal runaway protection since I have tested that already, and there is no warning screen displayed in this case.

The issue is not exhibited when the printer is started untethered or when the host is connected after boot. Looking at the reference to the workaround posted by @Msq001 in https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/issues/2#issuecomment-513446603_ I'm not seeing the ```//#define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE``` in the ```pins_BIGTREE_SKR_E3_DIP.h``` and when manually defined, will compile without issue, but it does not have any effect. The same behavior remains.

Is there a possibility a similar workaround can be implemented with this board, or is there a fix for this prblem being worked on across all the new line of boards right now? Any help is appreciated.

bigtreetech, 2019-08-03:
Does the temperature of bed and nozzle -14C?  If only hotend is -14C and the bed temperature is normal, it may be a problem with thermistors or motherboard.   

In fact, the problem with Xenon-2510 only happens when the USB CDC is connected through printrun at startup, and then it will report errors directly and can't do anything.

Please confirm that hotend temperature is normal without USB CDC connection, and if it is normal, it may be a bug in firmware. If it's also abnormal, it may be a hardware problem.

reloxx13, 2019-08-03:
The printer shows current 72°C E0 and -14°C HB after boot with USB by a roomtemp of 21°C

After unplugging USB and rebooting the printer the temp shows the room temp.

Ender3, SKR mini e3, same issue.

bigtreetech, 2019-08-03:
Actually. E3-DIP did not have this problem in my tests  https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/tree/master/Firmware. You can also download this to update mini-e3 (only for test this issue), and there it should not happen again

bigtreetech, 2019-08-03:
I will updated mini-e3 firmware to lastest version of Marlin. It should be ok.

pyunramura, 2019-08-03:
> Does the temperature of bed and nozzle -14C? If only hotend is -14C and the bed temperature is normal, it may be a problem with thermistors or motherboard.
>
> In fact, the problem with Xenon-2510 only [...]

The hotend temperature is normal when powered on without a USB host connected. The reading for E0 and HB are around 30c (it's hot here). In my case the problem presents itself only when USB is connected and printer powered on afterwards. It seems unlikely to be a hardware problem as I'm seeing normal values when untethered and test prints with the reused g-code are the same as before the board swap, so it leads me to believe the thermistors for E0 and HB are reading the same values as on the stock Melzi board. I'll give `#define BOGUS_TEMPERATURE_GRACE_PERIOD` a try along with flashing the referenced firmware. Thanks.

pyunramura, 2019-08-03:
I have done some testing on all the mentioned things. `#define BOGUS_TEMPERATURE_GRACE_PERIOD` and `#define THERMAL_PROTECTION_GRACE_PERIOD` have had no effect on the issue. I have also flashed the pre-compiled binary, in this case "firmware_tmc2208_chip_on_bottom.bin" in the first case, then cloned the mentioned Marlin repo and compiled with the only change specifying TMC2208 in `Configuration.h`. Both have led to inconsistent results. 

**Pre-compiled binary**

1. After flashing the pre-compiled binary on untethered power on, E0 is reading as expected.

2. When tethered powered on with printrun and connected manually after boot, E0 is reading as expected. 

3. When tethered power on with printrun manually connecting during boot process, one of two outcomes.
a. When attempted connection during early boot, buzzer is triggered continuously and Marlin fails to boot into Info Screen.
b. When connection with printrun is started manually during late boot the Info Screen is reached with no buzzer, E0 is reading -14c. The same behavior persists.

4. When tethered power on with OctoPrint the buzzer is triggered continuously and Marlin fails to boot.

**Cloned repo compile (changing driver only)**

1. After flashing the user compiled binary on untethered power on, E0 is reading as expected.

2. When tethered powered on with printrun and connected manually after boot, E0 is reading as expected. 

3. When tethered power on with printrun manually connecting during boot process (early or late), E0 is reading as expected. 

4. When tethered power on with OctoPrint the Info Screen is reached with no buzzer, E0 is reading -14c. The original behavior is presented.

Venares, 2019-08-08:
Also noted on my own board yesterday.

opcow, 2019-08-10:
I'm seeing this issue as well. If the board is connected to my PC running Windows 10, I get a reading -14C on the hotend or both the hotend and the bed. If the board is connected to a PC or Raspberry Pi running Linux, I get a runaway heating error. Using BOGUS_TEMPERATURE_GRACE_PERIOD to give the board time to get to the info screen before it halts allows me to see that the hotend reads 300C on booting and then drops to 298C and stays there until the board halts).

The binary `firmware_tmc2208_chip_on_bottom.bin` dated July 25 from the BIGTREETECH github resolves the issue when connected to my Windows machine. However, the issue of a false runaway heating error persists when connected to either a PC or Raspberry Pi running Linux.

Jakobchr, 2019-08-14:
I also have the same issue, and for now the only work around is turning the printer on WITHOUT usb connected, and then connect it afterwards for OctoPrint to work.

I have also tried the precompiled firmware, stock marlin with configuration changes and this repository with only change to 2130 stepper drivers. But then i get the Thermal Runaway warning or just -14C on the hotend.

tilllt, 2019-08-15:
Same here, definitely a Bug.

bigtreetech, 2019-08-22:
look here https://github.com/MarlinFirmware/Marlin/pull/15026, we have fixed the usb cdc connection bug of  temperature. and have update source code

Jakobchr, 2019-08-27:
This fix is working for me.

pyunramura, 2019-09-03:
I've since switched to a SKR 1.3 board due to a problem with the BTT 2209 drivers, but since the fix seems to be working for others I'll go ahead and close the issue and reopen if the problem persists.

closed 2019-09-03 by pyunramura