bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #5: Supported serial baud rate over USB.

pyunramura, 2019-08-02:
Less of an issue and more of a question and to provide a future reference on these boards. What is the maximum possible baud rate for the Mini E3/ DIP over USB serial? I'm aware that there are external factors that can affect negotiation speeds of the serial connection, but what I'm wondering specifically about is the maximum supported by the board. 

The default configuration is set at 115200 and seems to work fine for most actions. I have set it as high as 250000 in firmware which compiles fine and is listed in the printer info but I am unable to connect at the 250000 from any of the hosts that I've tried with it, so I'm not certain if it is a cable quality issue or if the increased baud rate is not actually being honoured. Bring this up because a baud of 115200 can severely hinder printing performance when printing at high speeds, or high geometry models from a connected host; exhibiting pauses during the print as the buffer empties since the serial connection can't keep up. If 250000 is supported by the board it would be very beneficial for those use cases. Thanks for the help.

tmar88, 2019-09-19:
I have 250000 working and Octoprint is connecting fine at that rate. The locations I remember setting it were #define BAUDRATE 250000 within Configuration.h and monitor_speed = 250000 under the env section for your defined board in platformio.ini (may be others I do not recall though...).

brianredbeard, 2019-09-29:
In the context of the baud rate you might have better luck with 230400.  The use of 250000 is related to [challenges with Arduino as a µCU](https://github.com/MarlinFirmware/Marlin/issues/205) as opposed to using one of the [standard UART baud rates](https://electronics.stackexchange.com/questions/9264/what-standard-uart-rates-are-there).

Since the [STM32F103RCT6](https://www.st.com/en/microcontrollers-microprocessors/stm32f103rc.html) is using UART (aka serial) in hardware (albeit over the USB connection) it _might_ work better with standard baud/bit rates.

StefanBruens, 2019-12-20:
For any board which uses the native USB connection of e.g. the STM32, baudrate is completely irrelevant - the connection always uses the USB Full Speed rate (i.e. 12 Mbps half-duplex).

Only for e.g. AVR boards which have an USB-serial converter (CH34x, FTDI, PL2303, ...) which connects to a UART of the MCU, the baudrate matters.

oldman4U, 2020-06-08:
Hi @pyunramura.

Do you still need this ticket? Please close it if not.

Thank you

pyunramura, 2020-06-11:
Done, Thanks for the info.

closed 2020-06-11 by pyunramura