bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #6: Board fan 1 controllable in software

pyunramura, 2019-08-02:
I have both of the fan headers on the board connected to two of the three fans on the Ender 3, though only one of them seems to be able to be controlled through software. Is the fan1 header able to be turned on/off or controlled via PWM?

When the board was first installed, I had the hot end fan connected to the power header, mainboard fan connected to fan0, and parts cooling fan connected to fan1, as it appeared on the stock Ender 3 mainboard. On printer start the hot end fan (power header) is running at full which is expected. The board cooling fan is off, and controllable through the printer or via host, and the parts cooling fan is running at full with seemingly no way to turn it off or adjust it in software.

Looking at the pin reference sheet, there seems to be no software controllable pin on fan1 header, only listing it as "GND", is this possible to have it controllable via software, or would that ability only come with a hardware revision?

I've since switched the parts fan into fan0 header and board fan into fan0 header as an aside. I may have to resort to a hardware hack if we can't get this sorted. Thank you for any help.

bjoernwedi, 2019-08-02:
had the same issue like pyunramura: before the usb patch fan0 was controlled, afterwards fan1.
So if both pins are controllable, I would be interested to to power off the extruder fan when not heated and controlling the part fan separately.

pyunramura, 2019-08-05:
@bjoernwedi Do you know which commit the change came about? I'm trying to look through the code changes and not seeing anything obvious.

bjoernwedi, 2019-08-12:
Do not  anymore. I think it was in the period of early days of the btt e3dip repository in contrast to the original marlin one. But iighz be wrong.
Myself, I did already some research, but was also not successful.

GizmoTheGreen, 2019-08-12:
Looking at board pins file and board pinout. I assume Fan0 is for part cooling fan and fan1 is permanent ON. it goes direct to Voltage and GND.

Alphaelectric, 2019-08-12:
Good to find this discussion....I am having the same issue and was wondering if the Fans can simply be switched?

pyunramura, 2019-08-14:
@GizmoTheGreen That's how I have it connected right now. fan0 is PWM controllable and fan1 seems to be always on.

@Alphaelectric Yes they can. I have parts cooling fan in fan0 slot and hotend fan in fan1 slot, that way the parts cooling fan is  controllable. It would be ideal though if both were able to be controlled through software.

georgekucher, 2020-03-08:
Hi guys!
Faced the same issue today((
So is there a chance to enable another fan control?

donjakobo, 2020-05-11:
Has anyone found a solution for this? Is it a hardware limitation? My 'case fan' on fan1 is always on, and would love to keep it lower speed to reduce noise as it doesn't have to be 100% to still cool the step drivers.

GizmoTheGreen, 2020-05-11:
Personally I have a voltage regulator fed off the permanent fan1 to 12v to feed case and hotend fan at slower speed. But your mileage may vary.

oldman4U, 2020-06-12:
Hi. This board is a copy of the original Melzi board used in the Ender-3 (only with better components) and therefore offers exactly the same functionality. This includes one controllable fan and one which turns on as soon as the board is powered on.

@pyunramura Seems this ticket is not needed anymore, so could you please close it.

Thank you

closed 2020-10-09 by pyunramura