bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #8: Ender 2 Stock Display stays blank

Rushmonk, 2019-08-06:
Dear All,

I'm using the SKR E3 DIP on my Ender 2, working all fine except for the display. It stays blank and I can't change it to a mini display. Could you please get some support for this?

kleinerbiber, 2019-08-15:
In theory it could be used with the following pin configuration when you define it as MINIPANEL instead of CR10_STOCKDISPLAY, but I can't get it to work either. (I don't know if the bottom part is needed because I can't check it.)
Maybe somebody from @bigtreetech could look into this with my information given?

```
/**
 *          MINIPANEL (Ender-2)
 *                 _____
 *     (SCK) PA15 | · · | PB6 (BTN_ENC)
 * (BTN_EN1)  PA9 | · · | RST
 * (BTN_EN2) PA10 | · · | PB9 (LCD_CLK)
 *  (LCD_CS)  PB8 | · · | PB7 (MOSI)
 *            GND | · · | 5V
 *                 -----
 *                 EXP1
 */

  #define DOGLCD_SCK     PA15
  #define DOGLCD_MOSI    PB7
  #define DOGLCD_CS      PB8
  #define DOGLCD_A0      PB9

  #define BTN_ENC        PB6
  #define BTN_EN1        PA9
  #define BTN_EN2        PA10

  // Alter timing for graphical display
  /**
   * #ifndef ST7920_DELAY_1
   *   #define ST7920_DELAY_1 DELAY_NS(125)
   * #endif
   * #ifndef ST7920_DELAY_2
   *   #define ST7920_DELAY_2 DELAY_NS(125)
   * #endif
   * #ifndef ST7920_DELAY_3
   *   #define ST7920_DELAY_3 DELAY_NS(125)
   * #endif
   */
```

tobybonhof, 2019-10-20:
I want this aswell! Any updates?

kleinerbiber, 2019-10-20:
> 
> 
> I want this aswell! Any updates?

Support from BTT for their products = -1

tobybonhof, 2019-10-20:
Yeah doing a huge file dump on github and calling it opensource.... yeah right

choco107, 2019-10-26:
Any updates on this, need it also

kleinerbiber, 2019-11-30:
Ender-2 Display seems to be supported now (defined as `MKS_MINI_12864`)! - Too bad I already sold my board again.
[https://github.com/MarlinFirmware/Marlin/pull/15924/](url)

choco107, 2019-11-30:
Nothing shows on that link ??

kleinerbiber, 2019-11-30:
> Nothing shows on that link ??
Sorry, fixed it. URL wasn't linked properly.

choco107, 2019-11-30:
Aw great , does merged mean it's in the latest bugfix download ??

kleinerbiber, 2019-11-30:
> Aw great , does merged mean it's in the latest bugfix download ??

Yes!

choco107, 2019-11-30:
Fantastic I'll try it tomorrow

oldman4U, 2020-06-08:
Hi.

What is the status of this issue? 

@Rushmonk Could you please close this ticket in case you do not need it anymore.

Thank you

oldman4U, 2020-09-29:
Please help the community and close the ticket once you do not need it anymore. Thank you

kleinerbiber, 2020-10-07:
> @oldman4U 
> Please help the community and close the ticket once you do not need it anymore. Thank you

Please stop annoying everybody by spamming this message, thanks.

oldman4U, 2020-10-07:
Please help me to make this ticket system useful by getting rid of unneeded tickets and comments like yours.

Thank you

PS. You can delete all your comments in this ticket to get no notifications anymore.