bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #11: firmware.bin not renamed / deleted

tilllt, 2019-08-15:
When updating the firmware with a firmware.bin from sd-card, it stays there. 

So every time the printer is started it seems to flash the firmware, which is unnecessary I guess. 

A workaround is to delete the firmware.bin manually but I guess it should be done automatically through the update process.

opcow, 2019-08-16:
I've seen this behavior as well. I formatted the sdcard and not it is working as intended again.

pyunramura, 2019-08-16:
I've noticed the same issue. I've recompiled and flashed probably dozens of times on this board so far and the same behavior is observed. The firmware.bin will be flashed by the bootloader won't be removed from the card. On a second power cycle the firmware will be flashed again from the card, but then it will be deleted. It's just odd that it takes two times for it to be removed, but it hasn't been a big issue for me. I have also confirmed that the firmware has in fact been flashed after the first power on.

oldman4U, 2020-06-12:
Hello tilllt.

Seems this ticket has not been used for a while, so could you please close it.

Thank you

penright, 2020-07-21:
I am seeing this also, any solutions?

oldman4U, 2020-07-21:
Hi Paul.

Does it upload the firmware ?  Which size and format does the sd card have
which you use for the update?

penright, 2020-07-21:
There are a few moving parts to those questions. Let me try and answer it this way. Let me know if I am understanding where you are going.

When it was uploading the firmware it never did rename it. I first started with the bin from the BT GitHub. Then I started compiling with VSCode since I was trying to get a BLTouch working. I did get it to compile and flash, but as stated never renamed. I would just pull the SD card after flashing. Then you jump forward a few days and I wanted to change my grid from 3x3 to 5x5 and after that, I could not compile. So a friend got an E3 mini working by going to marlin 2.0 bug fix. So I downloaded that, whereas before I was using Marlin in the BT GitHub. I could not get it to compile using the header from the BT GitHub, so I used the Reddit he used for the E3 mini headers (2006)  and did a Windiff to the BT GitHub E3 Dip (2005) and got it working. My board did consume the bin, but still did not rename. 

Now I have an issue, I believe unrelated to this issue, where the board is not reading the SD card. That is why I said there are moving parts to my answer. While the board was reading the SD card, it was not renaming the bin. Until I get my board reading the SD card, I will not be able to test. I will repost when it is working so if there is anything you need me to try, let me know.

I think brick the board. The blue is not flashing and the LCD is black. I have posted on the facebook page without any response. Should I open a ticket here? I found the #2 ticket but I am hoping to find an easier fix.

oldman4U, 2020-07-21:
I would not care if it is renamed or not. The update seems to work and this
is important.

Usually the update or renaming fails because the SD card is bigger  16GB or
not formatted using FAT16/32.

penright, 2020-07-21:
You are correct, the update does seem to work when there is a bootloader. :-)

The SD card could be the issue, it is the 128meg one that comes with the board. While searching for how to fix my bootloader I came across this post and was curious if anyone had fixed it. Since I use Octoprint other than flashing the SD does not have to be in. I have ordered a new board next day, until then I will see if I can reload the bootloader. I do have an STLink.

oldman4U, 2020-07-21:
Sorry but i can not help with the bootloader.

BTT even recommends max 8GB. Hard to find such small cards these days😁

wisesokol, 2020-07-23:
After the firmware is completed, you need to wait a couple of minutes. You cannot press the encoder. 90 percent of the time the file is renamed to .CUR

oldman4U, 2020-07-23:
@tilllt 

Do you need any further information? Please let us know.

oldman4U, 2020-08-26:
@tilllt 

Please close this ticket

closed 2020-08-26 by tilllt