bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #12: TMC2209 Drivers in UART causes homing to crash and stall steppers

pyunramura, 2019-08-20:
I just received and installed my BigTreeTech TMC 2209 Drivers for X/Y/Z/E in UART mode and defined the 2209 drivers in this branch of Marlin. During homing, the X and Y axes do not register the endstops of either axis which causes the steppers to crash and stall against the endstop switches. I do not know what could be causing the problem since the only change made from my previously working config with TMC 2208 drivers was defining 2209 in Configuration.h and swapping the driver boards. Any help with this?

salacpavel, 2019-08-26:
Just to let you know there is a workaround, but you have to desolder DIAG pin from driver board. I have had the same issue, which I have escalated to BTT a week ago, no real response so far.

sicdigital, 2019-11-02:
Is the work around the same for v1.2 drivers?

frdmn, 2019-11-19:
@sicdigital Did you figure out if v1.2 is affected by this as well?

sicdigital, 2019-11-20:
@frdmn - I couldn't get the e3 dip working with the 2209s for this reason. I returned the drivers and just got the skr mini.

georgekucher, 2020-03-06:
Having the same issue on TMC 2209 v1.2
Board BIGTREETECH-SKR-E3-DIP-V1.1

nicolube, 2020-03-07:
Having the same issue on TMC 2209 v1.2
Board BIGTREETECH-SKR-E3-DIP-V1.0

sustmi, 2020-03-14:
Hi @georgekucher and @nicolube .

It looks like both BIGTREETECH-SKR-E3-DIP-V1.0 and BIGTREETECH-SKR-E3-DIP-V1.1 have the same wiring where DIAG pins on X/Y/Z stepper drivers are directly connected to the appropriate X/Y/Z-STOP pins of the STM32 MCU.

As TMC 2209 does not allow to disable output on the DIAG pin, you cannot have connected both end-switch and the DIAG pin at one time. If you want to use the end-switches connected to the X/Y/Z-STOP pins you need to desolder or cut the DIAG pin from the driver board.

See details in my reply in this issue: https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/21#issuecomment-599044069

Rikthepixel, 2020-05-15:
You will not believe the thing i just found! I think that inside of marlin the DIAG pins are not defined to the processor. I did a continuity check and turns out that the Stallguard pins are seperate from the endstop pins. The MKS TMC2209 which i am in possesion of has the diag pin on the side most nearest to the Vref adjustment screw

nicolube, 2020-05-15:
Int the touleshooting part you cand find this:
https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf
There is the solution to the endstop problem.
And also my bord has a diffrend processor als named in the documentation...
So every time i had restarted the printer the firmware was gone.
Now i have the promlem that my EEPROM keeps reseting it self.

georgekucher, 2020-05-15:
> So every time i had restarted the printer the firmware was gone.
> Now i have the promlem that my EEPROM keeps reseting it self.

Had similar issue.
If I remember right, I've disabled (commented) some feature to fix it.
But don’t remember which exactly((

sdaitzman, 2020-06-20:
@georgekucher can I see your firmware (config and config_adv) so I can diff it with mine?