bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #13: while homing z bltouch doesn't always stop the z axis

Ofloo, 2019-08-20:
Sometimes it doesn't respond to the BLTOUCH at all, and just keeps trying to go down.

Then when I turn the printer of and on everything works fine, .. this only happens sometimes, .. but it happens. For my ender3 it made a hole inside the plastic bed.

Usually it's proceeded by some probe errors. While probing the bed.

ZeroTrousers, 2019-09-07:
Per a post on the BIGTREETECH SKR MINI E3&E3 DIP User Group someone put the 5v rail on an oscilloscope and found there was a high degree of ripple (noise), with voltage dropping as low as 1.8V. 

This is either due to a poorly sized 5V converter or insufficient decoupling - or both.  

There is a purposed DIY fix - adding a 100uF capacitor (electrolytic, 10V or above) to one of the max-endstop positions and a .1uF or 1uF ceramic or Tantalum low ESR capacitor to the 5V and GND pins on the Red/Brown servo portion of the header to decouple the load and smooth the ripple. 

It's currently believed that the ripple is caused by interference from the stepper motors. 

There have been a number of posts from SKR Mini 1.1 users saying that this has fixed the reliability issues with the BL touch. 

There has been no response from BTT regarding this yet. I believe this will probably be fixed in the 1.2 version of the board - along with the issue of the DIAG pins for stallguard disabling endstops on the 2209's.

lococnc, 2019-09-08:
I installed my SKR Mini E3 DIP with 2208 drivers. Everything worked perfectly, I spend many hours ensuring I had correct settings in the firmware.

I had read of people having problems with bltouch and crashing into the bed.

Well today I had that happen, and I KNOW EXACTLY WHY. I had also decided today to cut off the 5v line from my octopi USB input. IMMEDIATELY upon doing that the bltouch started failing. So I put the 5v back from the USB port and everything is back to normal.

So, it is OBVIOUS there is a flaw in the 5v supply on this board. If anyone is having the bltouch crashing issue, plug in a USB cable to stabilize the 5v supply.

So I agree, the issue is a hardware one and is on the 5v supply line.

ZeroTrousers, 2019-09-09:
Thanks for the input. 

I'm currently looking at the best way to decouple the servo 5v and GND. 

The 5v line controls the deploy and retraction of the servo (Red/Brown wires) and if you put the pins onto the 3.3v line it will fail to function correctly too, as a result of low voltage - the pin will deploy but does not have the required current/voltage to reliably retract.  

As we know, the mechanism is very sensitive to interference from fans and heaters and will cause premature triggering. Bad supply voltage or line noise totally makes sense as causing the pin to fail to retract and trigger the stop properly, it messes with the magnetic field that the probe relies on.

sustmi, 2019-11-03:
I would like to notify you that there was a software bug that caused BLTouch communication failures because the _SERVO0_ signal was kind of mixed with fan PWM signal.
The issue was fixed here: https://github.com/MarlinFirmware/Marlin/pull/15547 .
@Ofloo, could you verify whether it fixes your issues?

oldman4U, 2020-06-08:
Hello Ofloo.

Could you please close this ticket.

Thank you

closed 2020-06-09 by Ofloo

SrGeek, 2020-06-17:
What is the value of the diag1 pin on a SKR-Mini_e3_dip?