bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #17: E0_AUTO_FAN_PIN

SchwalenbergJe, 2019-08-24:
Would like to enable E0_AUTO_FAN_PIN in the software to let the hot end cooling fan only run when its over a specific temperature (default is 50°C) but I can't get it to work.

Setting it to "FAN_PIN" (as its seems to be configured in the pins_BIGTREE_SKR_E3_DIP file) only throws a compilation error. Also the specific pin value (PA8) is not working.

Any suggestions?

nicstoun, 2019-08-26:
Yes same to me , interesting board 
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1388:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

Chaotnix, 2019-08-28:
If you set E0_AUTO_FAN_PIN you should comment out the #DEFINE FAN_PIN statement.
You can not set two pin assignments to the same pin.

nicstoun, 2019-08-29:
> If you set E0_AUTO_FAN_PIN you should comment out the #DEFINE FAN_PIN statement.
> You can not set two pin assignments to the same pin.

What is mean ? we comment the pin from DEFINE FAN_PIN(pa8)  ..but its not pin for this on FAN0..that is problem in this board. E0 FAN go to -+ directly ..

Chaotnix, 2019-08-29:
Ok, you can not control FAN1, because it has no PWM-pin. The only way to control FAN1 would be to connect the GND to an PWM-pin.  But I dont know, if any of the other PWM-pins can handle this because of the high voltage (24v vs 5v) and the load.

If you DEFINE E0_AUTO_FAN_PIN to FAN_PIN it would not control the hotend fan (FAN1). Instead it would control FAN0.

TimoBre, 2019-10-03:
@Chaotnix 
I don't understoud your advice :-(
The Pins are declared in PINs_BigTree_SKR_E3_dip.h  so :
#define HEATER_0_PIN         PC8   // EXTRUDER
#define HEATER_BED_PIN     PC9   // BED
#define FAN_PIN                   PA8

[SKR E3-DIP-PIN.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3687122/SKR.E3-DIP-PIN.pdf)

The Pin-Layout is crystal clear . PA8 = (FAN 0) == Extruder Fan

**How switch it now so, that the extruder fan is switched on above 50°C and not always?**

You said:
_PINs_BigTree_SKR_E3_dip.h_
// Heaters / Fans
//
#define HEATER_0_PIN       PC8   // EXTRUDER
#define HEATER_BED_PIN     PC9   // BED
//#define FAN_PIN            PA8

and
_Configuration_adv.h_

#define E0_AUTO_FAN_PIN FAN_PIN 
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
#define CHAMBER_AUTO_FAN_TEMPERATURE 30
#define CHAMBER_AUTO_FAN_SPEED 255

But noting changed in the behavior. What I dont understoud? 

Thank you!

myudshlihe, 2019-10-04:
@TimoBre 
FAN_PIN is for model cooling. Not for extruder fan. And this is located on FAN0 connector. 
FAN1 connector connected directly to power source. So there is no way you can control it by software.

TimoBre, 2019-10-05:
Fan0 is controlable. see SKR-E3-DIP-V1.0-SCH.pdf. Its a logic IC behind.
So the question is, how to adress PA8, so that I can use him with.
Behind FAN0 ist a OptiMOS®3 Power-Transistor as an switch it is controled
by a CMOS MC74HCT125A. Its the same locic like the SKR 1.3 board and there it works.

myudshlihe, 2019-10-05:
Yes. FAN0 is controllable. But it is for model cooling. If you make it extruder cooler you will loose model cooling. Because second fan connector (FAN1) is connected directly to power supply. If you do not need model cooling and just want to have controllable extruder fan, I can try do this Monday.

myudshlihe, 2019-10-05:
I used trigorilla 8bit board before this, and there was 3 controllable fan connectors. It was great. 1 for extruder cooling, controlled by extruder temperature, 1 for model cooling controlled by slicer, and 1 for motor drivers cooling activated when motors active. On this board we have just 1 controllable fan, and this is not so fun.

TimoBre, 2019-10-05:
hey, you're right. I'm sorry, my brain was missing :-)

closed 2020-10-01 by jxns