bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #18: TMC5160 connection error

n-e-g, 2019-08-29:
I have problem using SKR-E3-DIP-V1.0 with TMC5160 drivers.
Configuration includes
```
#define X_DRIVER_TYPE  TMC5160
#define TMC_USE_SW_SPI
```

Yet TMC seems not to be responding:
```
SENDING:M122
		X
Enabled		false
Set current	800
RMS current	790
MAX current	1114
Run current	23/31
Hold current	11/31
Global scaler	130/256
CS actual	31/31
PWM scale	4294967295
vsense
stealthChop	true
msteps		0
tstep		4294967295
pwm
threshold
[mm/s]
OT prewarn	true
OT prewarn has
been triggered	false
off time	15
blank time	54
hysteresis
-end		12
-start		8
Stallguard thrs	0
DRVSTATUS	X
stallguard		*
sg_result	1023
fsactive		*
stst		*
olb		*
ola		*
s2gb		*
s2ga		*
otpw		*
ot		*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
```

I there any special configuration needed to use TMC5016 with this board ?

hopatz, 2019-12-06:
I have skr e3 dip with 4 tmc5160 v1.2 and no problems.. maybe you need try with newest bugfix marlin 2.0

RedTornado66, 2020-03-05:
Did you connected 24V during test?

oldman4U, 2020-06-12:
Hello n-e-g.

Please close this ticket.

Thank you

oldman4U, 2020-09-29:
Please help the community and close the ticket once you do not need it anymore. Thank you

n-e-g, 2020-09-29:
Not needed anymore

closed 2020-09-29 by n-e-g