bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #21: Endstops do not function when DIAG pins of BTT TMC2209 V1.1 are plugged in.

ZeroTrousers, 2019-09-07:
The board does not function as intended when BTT TMC2209 V1.1 drivers are inserted. Endstops will fail to trigger in all cases. 

If the pins are removed or not inserted, endstops function normally. 

No settings in Configuration.h or Configuration_adv.h will cause the endstops to trigger if these pins are inserted. 

The drivers do not report a full M122 without the pins inserted.

sustmi, 2019-09-09:
Is this a duplicate of https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/12 ?

sl1pkn07, 2019-09-11:
seems no wire route to any pin. dead pinheader

smolz, 2019-11-06:
What is the fix?  don't use 2209's?

sl1pkn07, 2019-11-06:
wire the diag1 pin with wire directly tlo endstop pinheader

smolz, 2019-11-06:
So does that mean removing my existing end stops and turning on the senseless homing?

sl1pkn07, 2019-11-06:
if you want use sensorless, yes, if you use physical endstop, then you need remove the diag1 pin in the driver

smolz, 2019-11-06:
Removing now

smolz, 2019-11-06:
what about the other pin that is next to it?  leave that? also, it looks like I am supposed to leave one of the jumpers in under the driver?  Is that true information?

georgekucher, 2020-03-06:
Having the same issue on TMC 2209 v1.2
Board BIGTREETECH-SKR-E3-DIP-V1.1

domhardt, 2020-03-14:
> Having the same issue on TMC 2209 v1.2
> Board BIGTREETECH-SKR-E3-DIP-V1.1

I have the same problem. I'm compiling Marlin 2.0.4.4 for BTT E3 DIP (STM32F103RE_bigtree) with TMC2209 over UART. Everything is working fine except the mechanical endstops. The micro switches are always open and do not trigger even when shorted out. What are the magic options to have TMC_DEBUG (M122) and STEALTHCHOP but use the stock endstops (no sensorless homing)?

sustmi, 2020-03-14:
Hi @domhardt .

Neither the official [schematic diagrams](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/blob/master/Hardware/SKR-E3-DIP-V1.0-SCH.pdf) nor the [pin description diagram](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/blob/master/Hardware/SKR%20E3-DIP-V1.0-PIN.pdf) show how the DIP pins are connected.

Here you can at least see the description of the pins of the DIP connector: 
[image]
(source: https://www.biqu.equipment/products/bigtreetech-tmc2209-stepper-motor-driver-for-3d-printer-board-vs-tmc2208?_pos=1&_sid=a34306611&_ss=r )

Using my multimeter on SKR-E3-DIP-V1.0 I can check that for example DIAG pin in the DIP connector for Z-axis driver is directly connected to pin 4 of the STM32 MCU which is Z-STOP sense pin. That means that if you do not desolder or cut the pin on the driver board like I or [@salacpavel did](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/12#issuecomment-524918420) those two signals (endstop and DIAG) will be in conflict.

Actually, the Z-STOP signal from the endstop switch is passed to the MCU through 1k resistor while the DIAG pin from the driver board is not (see https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/Schematic/TMC2209-V1.2.pdf ) so the endswitch signal is overriden by the DIAG output signal.

If you want to use the endstop switches connected to (X/Y/Z)-STOP pins, you need to cut or desolder the diag pin from the driver boards. BTT even has this in their troubleshooting manual:
> the DIAG of the TMC2209 can only be used as an output. Therefore, it should be noted when using TMC2209 on the board with DIAG function :(such as SKR E3 DIP V1.0, BTT002 V1.0, SKR Pro V1.1, etc.) If you use the TMC2209 but needn’t use the DIAG function, you have to cut the DIAG pin on the TMC2209 driver, otherwise the limit switch function will not work properly.

(source: https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf )

closed 2022-01-30 by ZeroTrousers