bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #23: EEPROM not loading on startup.

GForce2010, 2019-09-13:
I have enabled EEPROM in configuration.h

Using the LCD I can save setting to the EEPROM but when I power cycle the printer the settings don't return unless I manually load setting from the LCD panel.

What do I need to do it get it to load automatically as the Marlin wiki says it should.

Thanks.

federico77, 2019-12-19:
Having the same issue. Do you knpw if this is fixed or has fixes?

AalianKhan, 2020-02-09:
Apparently there was a bug in Marlin and it has been fixed in the latest version of vanilla Marlin, but bigtreetech didn't yet come around to fix the problem. While searching for a solution a fantastic redditor made a fantastically detailed post on what settings to change/enable/disable for the SKR mini e3 v1.2 to function. Here is the link to the [post](https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/)