bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #26: constant buzzer sound & no image on LCD

rzhvsk, 2019-09-21:
Hey guys!

I have encountered several problems and i am too desperate to solve them solely.
(e3-DIP v1.0 with 2208's in UART mode)

Pre-issue: 
After initial swapping of the board everything was almost fine. LCD screen was saying "TMC connection error", but despite this message all XYZ movements worked. Unfortunately, all movements were reversed (so e.g. i couldn't ever finish "auto home" process). No big deal, just couple tweaks here and there in Marlin; i've done that ~1000 times with other 3d-printers, right?

Issue:
After the very first firmware upgrade (i grabbed one pre-baked from here while vscode was downloading all prerequisites) via sd-card i got subject issue: there was no image on LCD, only backlight; and speaker/buzzer was screaming single tone non-stop.
Then i started flashing every firmware i could compile or find online, yet still this buzzer is slowly killing me. None of the advice found (defining this and that, flipping lcd connector 180 degrees, renaming pins in pins_..._...h) worked for me. I guess pins layout in firmware and on board do not match, but what do i know? I could add some pictures so wise men could tell me to sit down.

There is blue LED flashing on boot, "firmware.bin" gets renamed to "FIRMWARE" and shrunk in size to 0b, so i still see some hope that board isn't dead (but this means i'm just stupid).

My last attempt was in compiling Marlin 2.1.x (dev build), applying settings for SKR-E3-DIP (as this version has this board in list of default_envs and pins). Buzzer won't shut up even if everything is disassembled (besides lcd connector) and powered via USB.

Also, there is no connection via USB to chat with the board; there are no additional devices in COM-ports once i connect the board.

1. How do i make this board appear in devices among COM-ports? Do i have to add jumper somewhere?
2. My guts tell me this issue has to do something with lcd connector pins layout, how do i verify that and how do i deal with it?
3. If i purchase TFT screen from bigtreetech and wait another couple of weeks, could i replace stock ender 3 display without having to fight this frustrating issue?

I would appreciate your help and advice very much, thank you.

chaoticchoas, 2019-09-23:
Your board shouldn't be dead. It just seems like the bin file you tried to flash was corrupted somehow(maybe you unplugged the sd card while it was still transferring). The buzzer goes off while the board is flashing the firmware.bin from the sdcard. You should see a flashing blue light and a solid red light on the top of the board opposite of where the usb and sd card plug in. The blue light should turn solid when it boots the firmware. As for your tmc connection error it seems like you need to do [Step 2](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/tree/master/Firmware) if you are using the latest version of marlin but it will give you an compiling error if you do so you need to further edit the platformio.io file like on [my post to fix it](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/27). If you like I can just post my platformio.io.

rzhvsk, 2019-09-24:
Yes, I see red led (3.3v) working constantly. Blue led is flashing for 1 or 2 seconds, then it turns off (it also works constantly if i hold reset button).

For now i do not have any TMC connection errors as i do not have any info on the screen, but i will try compiling firmware with your fixes in case this magically solves everything (i really hope it does).

rzhvsk, 2019-09-24:
chaoticchoas, I compiled your entire fork of marlin, no changes in board behaviour :(

rzhvsk, 2019-09-24:
microsd card that came with the board was broken during shipping (physically damaged).
all of these tests were performed with... 8 gb microsd from ender 3.

I assume it's time for me to lie down and try not to cry..?

chaoticchoas, 2019-09-25:
Yea it seems like it until someone with more knowledge than me can help you. If you haven't already I would try and just reformat the card since it seems like [This Person](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/61l) Was having a similar issue with is skr 1.3.

rzhvsk, 2019-09-25:
> Yea it seems like it until someone with more knowledge than me can help you. If you haven't already I would try and just reformat the card since it seems like [This Person](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/61) Was having a similar issue with is skr 1.3.
> 
> Edit: fixed link

Yes, i tried two microsd cards (from ender 3 and the one i bought for raspberry pi), both 8 gb, firmware.bin gets renamed only with fat32 and empty "name" of the card; no luck yet. 

Thank you for your effort!

rzhvsk, 2019-09-29:
So, it's time for a conclusion.

I bought an st-link/v2 programmer and connected to e3 DIP board using SWD pins at its center. I was able to read firmware, compare it to firmware with bootloader from issue #2, erase existing and upload a new one. (STLINK32 Utility).

After that board completely "bricked", blue led was working constantly, buzzer and screen issues persisted, programmer 

Then I used "firmware.bin" for tmc2208 with chip on the bottom from repository, board successfully picked it up and booted, it appeared in USB-devices as maple. I was hopping on the carpet, feeling catharsis after two weeks of wild sex with the board.

My next journey lies deep within BLTouch sea of issues, wish me luck.

closed 2019-09-29 by rzhvsk

Poeb69, 2020-02-06:
Hi. Can you help me with ST LINK v2 progammer ? After what I connect board SKR E3 DIP, STLINK Utillity says : Read out protection is enabled. I try in Tools - Option bytes to select disabled, but ST LINK utillity still says : Read out protection is enabled. I don´t know, what i do ? Can you help me please ? Sorry, my english is not good.

Poeb69, 2020-02-06:
> So, it's time for a conclusion.
> 
> I bought an st-link/v2 programmer and connected to e3 DIP board using SWD pins at its center. I was able to read firmware, compare it to firmware with bootloader from issue #2, erase existing and upload a new one. (STLINK32 Utility).
> 
> After that board completely "bricked", blue led was working constantly, buzzer and screen issues persisted, programmer stopped working every time I connect it to the board.
> 
> Then I used "firmware.bin" for tmc2208 with chip on the bottom from repository, board successfully picked it up and booted, it appeared in USB-devices as maple. I was hopping on the carpet, feeling catharsis after two weeks of wild sex with the board.
> 
> My next journey lies deep within BLTouch sea of issues, wish me luck.

Hi. Can you help me with ST LINK v2 progammer ? After what I connect board SKR E3 DIP, STLINK Utillity says : Read out protection is enabled. I try in Tools - Option bytes to select disabled, but ST LINK utillity still says : Read out protection is enabled. I don´t know, what i do ? Can you help me please ? Sorry, my english is not good.

Poeb69, 2020-02-06:
> So, it's time for a conclusion.
>
> I bought an st-link/v2 programmer and connected to e3 DIP board using SWD pins at its center. I was able to read firmware, compare it to firmware with bootloader [...]

Hi. Can you help me with ST LINK v2 progammer ? After what I connect board SKR E3 DIP, STLINK Utillity says : Read out protection is enabled. I try in Tools - Option bytes to select disabled, but ST LINK utillity still says : Read out protection is enabled. I don´t know, what i do ? Can you help me please ? Sorry, my english is not good.

tonortiz01, 2020-02-09:
> Hi. Can you help me with ST LINK v2 progammer ? After what I connect board SKR E3 DIP, STLINK Utillity says : Read out protection is enabled. I try in Tools - Option bytes to select disabled, but ST [...]

I have the same problem

Poeb69, 2020-02-09:
I had succeed just a moment. ST-LINK says me : Read out protection is
enabled, again and again. I don´t know how, I set ST-LINK. I to give up
with ST - LINK. I try load firmware from bigtreetech github : firmware with
chip on bottom. I give firmware on SD card, turn on printer. Still blue
screen, BUT I connect USB, when 3D printer turns on, and board boot :) I
dont understand it. Problem is, when I turn off printer, disconnect USB,
again BLUE screen. I try the same process with USB and board boot. I turn
off 3D printer and again blue screen. Here I end. Try it, maybe you find
more. I will like, when you let me know. Again sorry, my english is not
good. I am from Czech Republic. Bye.