bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #29: Marlin seems to crash when the printer is teathered to the pc with cura opened.

chaoticchoas, 2019-10-03:
I am having a weird issue that whenever I have my printer tethered to my PC with Cura opened it seems to cause the printer to no longer respond to anything and a loud alarm like sound coming from the board until I reboot the printer and either close Cura or un-tether it. It so far seems like this is happening with only cura as tethering the printer without cura or using pronto face does not seem to cause the printer to freeze.

Steps to reproduce:

1. Turn on printer
2. Tether it to the computer
3. Open Cura
4. Wait for windows to recognize the device.
5. Printer Freezes and the alarm like sound comes from board.

PepekT, 2019-10-21:
I have similar issue.
 I tried to send few commands trough Pronterface. 
I am able to connect printer, send command but there is no response from printer.
Printer is also freezed (I tried to control printer trough encoder with no success).

b4tman, 2019-10-21:
Commenting out 5 lines (33-37) in [plugins/USBPrinting/AutoDetectBaudJob.py](https://github.com/Ultimaker/Cura/blob/master/plugins/USBPrinting/AutoDetectBaudJob.py#L33) should fix this.

https://github.com/Ultimaker/Cura/issues/6154