bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #30: Build error: Git Client VS/platformio

Reefsider, 2019-10-04:
Hi, I'm new to marlin so forgive me if this is a newbie question.  I did look online and couldn't find anything useful.  

I followed the instructions on here about changing the .ini file for the steppers.  The steppers is what the build gets stuck on.

When i go to "build" in platformio it tells me:

Processing BIGTREE_SKR_MINI (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.6.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing TMCStepper
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\reefs\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\reefs\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\reefs\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\reefs\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\reefs\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\reefs\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 122:
    _build_project_deps(env)
  File "C:\Users\reefs\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\reefs\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\reefs\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\Users\reefs\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\reefs\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 359:
    force=force)
  File "c:\users\reefs\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 701:
    track=True)
  File "c:\users\reefs\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 486:
    vcs = VCSClientFactory.newClient(tmp_dir, url)
  File "c:\users\reefs\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 49:
    silent)
  File "c:\users\reefs\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 63:
    self.check_client()
  File "c:\users\reefs\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 130:
    "Please install Git client from https://git-scm.com/downloads")
====================================================================== [FAILED] Took 1.50 seconds ======================================================================

So i went and downloaded and installed the items.  then tried again, same problem, so I re-started the computer.  Tried again, same problem.  I then tried to close all folders and windows and everything in Visual Code/ Platformio cose everything try again same problem.  My computer is up to date as is VS and Platformio.  I'm using marlin 2.0.  What am I doing wrong???

TimoBre, 2019-10-05:
Hi, Do you have DL the lastest Marlin 2.0 Version? It don't looks like. Maybe 2 weeks older.
If I look at the platformio.ini. I dont find an SRK MINI declaration anymore.
So DL the newst. Here you find the [env:STM32F103RC_bigtree_USB] or [env:STM32F103RC_bigtree] w/o USB. 
If you compile it now atom will say e.g.

Processing STM32F103RE_bigtree_USB (platform: ststm32; framework: arduino; board: genericSTM32F103RE).

rzhvsk, 2019-10-05:
> If I look at the actual platformio.ini. I dont find an SRK MINI declaration anymore.
> So DL the newst. Here you find now [env:STM32F103RC_bigtree_USB] or [env:STM32F103RC_bigtree] w/o USB.

I assume @Reefsider compiles firmware from this repository, so it should build normally.

Could you check nesting? I got all sorts of errors while compiling in /users/.../downloads. There is a recommendation on nesting somewhere here (I did in C:\mar\dip\ and it was ok)

Reefsider, 2019-10-05:
Hi, I have uninstalled VS code and reinstalled it.  moved the files to the C:\ drive and downloaded the latest version of marlin 2.0.   I am just finished editing the configuration.h file and updating the env in the .ini file.  

Do i still need to modify the TMCstepper thing (https://github.com/bigtreetech/TMCStepper) or do i just leave that alone now?

rzhvsk, 2019-10-05:
As far as I remember, it won't compile with TMCStepper@..., I did with bigtreetech github dependency
(I might be wrong)

Reefsider, 2019-10-05:
I tried it and I now get this message:

Marlin\src\lcd\extui_malyan_lcd.cpp: In function 'void process_lcd_j_command(const char*)':
Marlin\src\lcd\extui_malyan_lcd.cpp:191:3: error: expected ',' or ';' before 'switch'
   switch (command[0]) {
   ^~~~~~
Compiling .pio\build\STM32F103RC_bigtree\src\src\lcd\menu\menu_advanced.cpp.o
*** [.pio\build\STM32F103RC_bigtree\src\src\lcd\extui_malyan_lcd.cpp.o] Error 1
===================================================================== [FAILED] Took 81.42 seconds =====================================================================

Also wondering about a UI for my lcd.  I was planning to use the original MPSM V1 LCD but I do have a Bigtreetech TFT35 V2.0 I could use if its easier.

Reefsider, 2019-10-05:
I found the file it is referencing in the marlin/src/ folder.  I am not really sure how to edit it to work though.

first question is what serial port does the SKR mini E3 DIP use for the UI?

// On the Malyan M200, this will be Serial1. On a RAMPS board,
// it might not be.
#define LCD_SERIAL Serial1

is that correct?

Reefsider, 2019-10-05:
So it compiled finally.  There was a missing ";" in the code.  Once I figured out where the file was and what the error was referring to it seemed obvious but I have never done this before so it is a bit of a learning curve.  Thank you for your help it is really appreciated.

Reefsider, 2019-10-06:
ok so now i can't get the computer to recognize the board.  I'm sorry if this is basic stuff but I can't seem to figure it out and the instructions and board manual didn't help.

I downloaded the mapleDRV-USBdriver folder but i can't find an executable file.

I also tried a different USB cable but that didn't do anything.  Also tried plugging it into my imac and still nothing.  maybe something is wrong with the board?  the red and blue lights are on when i plug it in though

TimoBre, 2019-10-06:
HEy Reefsider. You had DL the latest Marlin?. If yes look at here https://github.com/MarlinFirmware/Marlin/issues/15254. You have to enable the USB drive in the code. Then it works. After that your PC will see the Board and SC Card :-) . Serveral Problems are discussed and solved there.  Set the filter e.g. to: " is:issue is:open skr e3 dip"

richf62, 2019-11-09:
> So it compiled finally. There was a missing ";" in the code. Once I figured out where the file was and what the error was referring to it seemed obvious but I have never done this before so it is a bit of a learning curve. Thank you for your help it is really appreciated.

Where exactly is the location of the missing ";"