bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #31: Skr e3 dip v1.0 + tmc2208 v3.0 stepping value error

hopatz, 2019-10-06:
Just install skr e3 dip v1.0 with tmc2208 uart v3.0 stepping all axis is problems.. default stepping x y 80 is too big.. moving manually x 100mm almost over the bed.
Try set x stepping value 40 seems normal.. i dont what happened.. maybe anyone can guide flash into my ender 3? Thanks

MacBraky, 2019-10-07:
same problem here. New skr E3 dip v1.0 with tmc2208 uart v3.0. 10mm move results in 20mm effective move. Using vanilla Marlin 2.0.X. Will try with bigtree marlin to see if it helps.
Using the precompiled firmware for "E3 DIP with TMC2208 bottom" works fine.

MacBraky, 2019-10-07:
It's an issue in marlin: https://github.com/MarlinFirmware/Marlin/issues/15012

hopatz, 2019-10-08:
yes pre compile firmware is fine but i need bl touch sensor :( 
"tmc connection error" is appears too

hopatz, 2019-10-08:
adding -Dhave_serial platform.io result blank lcd

hopatz, 2019-10-08:
https://github.com/teemuatlut/TMCStepper ---->> compiling OK but manul move 10mm run 20mm
  
 TMCStepper@>=0.5.0,<1.0.0 ---->> compiling OK but manul move 10mm run 20mm

and

https://github.com/bigtreetech/TMCStepper ---->compiling fail error

collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_bigtree\firmware.elf] Error 1

MacBraky, 2019-10-08:
Use the source of marlin on bigtree. The current source on marlin github is not stable. It works fine with bigtree source.

TheFoXi, 2019-10-08:
Hello
I also encountered this problem after installing the E3 DIP and 2208 UART and the latest version of marlin 2. The motor step was 2 times larger and the error "tmc connection error"
BigTree is very old and for the new hardware I want new software.

I did not know what to do and decided to compare the firmware from Bigtree and the original Marlin 2 and found that in version> 0.4.5 TMCStepper, an additional argument must be specified for compilation -DTMC_SERIAL_SWITCH.(https://github.com/teemuatlut/TMCStepper/blob/b5b3658d34f93e0cc646dbaa54f90b809267206f/src/TMCStepper.h#L26)

Something like this: (Line 305 in platformio.ini)
`[env:STM32F103RC_bigtree]
...........

build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py 
${common.build_flags} 
-DDEBUG_LEVEL=0  
-std=gnu++14 
**-DHAVE_SW_SERIAL** 
**-DTMC_SERIAL_SWITCH**
..............`

and this in lib_deps = (Line 28)
  TMCStepper=https://github.com/teemuatlut/TMCStepper/archive/master.zip
  #TMCStepper@>=0.5.1,<1.0.0

It works, but I don’t know how correct it is :)

junemars88, 2019-10-11:
If you want to install vanilla marlin 2.0 you have to change a few things in your platform.ini file. 

**under [platformio]**

change "default_envs = megaatmega2560" to "default_envs= STM32F103R_bigtree"

change  "TMCStepper@>=0.5.0,<1.0.0" to "https://github.com/bigtreetech/TMCStepper"

**under [env:STM32F103R_bigtree]**

Add the line shown below in the same spot you see shown.

  ${common.build_flags} -std=gnu++14
  -DDEBUG_LEVEL=0 -DHAVE_SW_SERIAL    <-------------ADD THIS LINE 
build_unflags     = -std=gnu++11

If you are using ATOM to compile your code make sure you select "Rebuild C/C++ Project Index." under the platformio menu before compiling. I found that this helped whenever I got errors. 

Make sure to configure you configuration.h file and it should compile.

aaronkt, 2019-10-14:
I'm a decent noob here. I'm also having the issue with my printer moving to far. And the tmc connection error. Also did you all use the jumper cables at all? I didn't plug hem into anything and it seems to work? I didn't get any install instructions so I went with what I thought was right. Thanks

chaoticchoas, 2019-10-21:
If you are using the BTT TMC 2208 3.0 you should follow this [picture](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/blob/master/Hardware/UART.png) in which jumpers to use. It is should on the second to bottom row and the jumper on only the 2 left pins which should be red.

igoronofr, 2019-11-01:
I have the same problem, move twice and TMC CONECTION ERROR in display.

Aleix-casa, 2020-01-08:
where can I find the corect filmware to fix this problem?

closed 2020-06-09 by hopatz