bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #33: Axes are misdirected and run in the opposide way

Robin2305, 2019-10-10:
I insalled the firmware for the TMC2208 (chip on bottom ) and try to run it (the firmware was installed and stardet up) then i tryed to move the axis but they are completle mistaken. that means when i want to move the X-axis the Y-axis moved and when i want to move the Y-axis the Z-axis moved (and so on). I definitely pluged the steppermotors in the right possition.

Second issue is that they are running the wrong way - if i go +10mm it moved -10mm.

Does anybody have the same problems ore does anybody know the solution fore my problems ?

chaoticchoas, 2019-10-10:
I would suggest quickly check your axis wiring to make sure that it is correct. Then you can try the other tmc2208 firmware (chip on top). If you are still having trouble with the axis moving in the wrong direction or the wrong axis moving then just build your own version of marlin 2.0.x bugfix and you should be able to fix it there.

Mesmer88, 2019-11-21:
What printer do you have?
I experienced the same on Ender 5 and had to use these settings in Marlin:

#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG

oldman4U, 2020-06-08:
Hello Robin2305.

Seems this ticket has not been used for a while, so could you please close it.

Thank you

closed 2020-10-01 by Robin2305