bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #37: Compiled for CR-10, Board does not show up as a USB device after installing, no LCD output

pyr0ball, 2019-11-09:
I'm able to install the pre-compiled binaries posted in this repo, but whenever I try to compile my own (using VSCode/PlatformIO on Windows 10) the board no longer responds. LCD stays lit but blank, no serial devices are detected on the raspberry pi I'm running octoprint on, nor on my linux laptop when I plug into that to test.

I've tried both the BTT fork with it's base configuration, with my own customized version for the CR-10, as well as the latest Marlin-2.0-bugfix branch.

Here's the pio config for the BTT fork:
```
[platformio]
src_dir      = Marlin
build_dir    = .pioenvs
lib_dir      = .piolib
libdeps_dir  = .piolibdeps
boards_dir   = buildroot/share/PlatformIO/boards
env_default  = BIGTREE_SKR_MINI
```

and the official marlin:

```
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC_bigtree
```

pyr0ball, 2019-11-09:
Issue seems to be related:
https://github.com/MarlinFirmware/Marlin/issues/15767

oldman4U, 2020-06-08:
Hello pyr0ball.

Seems this ticket is not needed anymore, so could you please close it.

Thank you

closed 2020-09-29 by pyr0ball