bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #40: Unable to connect to USB after upload of firmware

RyanParle, 2019-11-26:
So i'm really struggling with this board!

Now I cannot get a USB connection when I compile and upload the BTT-SKR-E3-DIP-V1.0 Firmware, i insert the SD card and press the reset button then the blue activity LED flashes for around 5 seconds but after that there is no USB connection made to the PC. When I check the SD card it still has firmware.bin in place and the blue LED will flash on every power cycle.

however if i compile a version of vanilla marlin 2.0 bugfix with just the board and serial ports set the board goes through the usual firmware load sequence and then connects to the PC as normal, the firmware.bin file has also been removed from the SD card. I would normally continue with the vanila marlin build but as soon as I enable the TMC 2130 features the compile fails as it is too big to fit on the ROM.

oldman4U, 2019-12-10:
Hi RyanParle.

The firmware provided by BTT seems to be broken and only the precompiled firmware packages are working fine. Fix seems to be on the way, please see the other tickets related to the same issue. Using the non BTT Marlin version also works.

oldman4U, 2019-12-18:
Good morning.

Have you been able to solve the reported issue? Please let us know.

Thank you

RyanParle, 2019-12-18:
I gave up with the BTT branch and moved over to the Vanilla Marlin Bugfix 2.0x.

As far as I am concerned this is issue closed.

closed 2019-12-18 by RyanParle