bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #41: Skr E3 Dip V1.0

Johnathan2581, 2019-11-30:
Hello,

I'm currently having an error that is stopping me from compiling my firmware for my 
Skr E3 Dip v1.0 with Tmc 2208 drivers. The Environment keeps saying it failed. When I go to Compile anybody knows what might be happening can't find much info on it. Sorry In advance this is just straight copy paste 

here's the error code:
Processing STM32F103RC_bigtree_USB (platform: ststm32; framework: arduino; board:
 genericSTM32F103RC)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC

.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.2, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), fram
ework-arduinoststm32-maple 1.10000.190819 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing U8glib-HAL
Using cache: C:\Users\johna\.platformio\.cache\a7\b7a9b473d1c239d9843ca943887cd0a
7
Unpacking...
WindowsError: [Error 206] The filename or extension is too long: 'C:\\Users\\john
a\\Downloads\\BIGTREETECH-SKR-mini-E3 bltouch updated standard mount\\BIGTREETECH
-SKR-mini-E3 bltouch updated\\Marlin-bugfix-2.0.x\\.pio\\libdeps\\STM32F103RC_big
tree_USB\\_tmp_installing-mmvgfl-package\\U8glib-HAL-bugfix\\src\\clib\\HAL_examp
le_files':
  File "C:\Users\johna\.platformio\penv\lib\site-packages\platformio\builder\main
.py", line 151:
    env.SConscript("$BUILD_SCRIPT")

  File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Scr
ipt\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Scr
ipt\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\johna\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Env
ironment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Env
ironment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool

s\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\lib
.py", line 301:
    force=force,
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\pac
kage.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\pac
kage.py", line 480:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\pac
kage.py", line 219:
    return fu.unpack(dest_dir, with_progress=with_progress)
File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\unpacker.py"
, line 145:
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\unpacker.py"
, line 39:
    self._afo.extract(item, dest_dir)
  File "c:\users\johna\.platformio\penv\lib\zipfile.py", line 1051:
    return self._extract_member(member, path, pwd)
  File "c:\users\johna\.platformio\penv\lib\zipfile.py", line 1102:
    os.mkdir(targetpath)

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED

megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      IGNORED
STM32F103RC_bigtree_USB  FAILED    00:00:13.700
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED

mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
==================== 1 failed, 0 succeeded in 00:00:13.700 ====================

wieltar, 2019-12-01:
WindowsError: [Error 206] The filename or extension is too long: 'C:\Users\john
a\Downloads\BIGTREETECH-SKR-mini-E3 bltouch updated standard mount\BIGTREETECH
-SKR-mini-E3 bltouch updated\Marlin-bugfix-2.0.x\.pio\libdeps\STM32F103RC_big
tree_USB\_tmp_installing-mmvgfl-package\U8glib-HAL-bugfix\src\clib\HAL_examp
le_files':
File "C:\Users\johna.platformio\penv\lib\site-packages\platformio\builder\main
.py", line 151:
env.SConscript("$BUILD_SCRIPT")

it could be your file path is too long, try moving the repository to the root of your c drive for example and try to compile it there

Johnathan2581, 2019-12-01:
I Will try that right away thank you

Johnathan2581, 2019-12-01:
Okay so I tried moving the repository to my direct c root folder and it has seemed to get rid of that error but I now have some more direct error I will also add a copy of my repository if anyone would like to look through it 
[Marlin-bugfix-2.0.x.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3908773/Marlin-bugfix-2.0.x.zip)
  

Processing STM32F103RC_bigtree (board: genericSTM32F103RC; platform: ststm32; fra
mework: arduino)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC

.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.2, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), fram
ework-arduinoststm32-maple 1.10000.190819 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing TMCStepper
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\main
.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\platformio.py", line 48:

    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Env
ironment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\lib
.py", line 301:
    force=force,
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\pac
", line 61:
    self.check_client()
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\vcsclient.py
", line 127:
    "Please install Git client from https://git-scm.com/downloads"

========================= [FAILED] Took 2.89 seconds =========================
 
Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
fysetc_f6_13                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       IGNORED
STM32F1_base                  IGNORED
STM32F103RC_base              IGNORED
STM32F103RC_fysetc            IGNORED
STM32F103RC_bigtree           FAILED    00:00:02.890
STM32F103RC_bigtree_USB       IGNORED
STM32F103RC_bigtree_512K      IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED

teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED4

wieltar, 2019-12-01:
The marlin script tries to get it's dependencies from the tmc drivers from GitHub, try to install the git client from: https://git-scm.com/downloads

wieltar, 2019-12-01:
I don't know if this is up your alley but I recommend installing the firmware from the main marlin 2.0 branch, it's updated (and keeps being updated, fixing bugs etc.) and fixed some of the issues with the BLtouch and it has options built in for the speculative "512k memory upgrade" 

More here:
https://www.reddit.com/r/BIGTREETECH/comments/dvgn4o/lies_the_skr_e3_mini_v12_actually_still_has_512k/?utm_medium=android_app&utm_source=share

Im planning on trying it out myself soon.

Johnathan2581, 2019-12-01:
Okay so I did install the git client and this is what I get now, It seems like the error is still present. I'll look into that "512K memory upgrade" it looks rather interesting.

Processing STM32F103RC_bigtree (board: genericSTM32F103RC; platform: ststm32; fra
mework: arduino)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC
.html

PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.2, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), fram
ework-arduinoststm32-maple 1.10000.190819 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing TMCStepper
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\main
.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Scr
ipt\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Scr
ipt\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
return self.method(*nargs, **kwargs)
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool

s\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\johna\.platformio\packages\tool-scons\script\..\engine\SCons\Env
ironment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\users\johna\.platformio\penv\lib\site-packages\platformio\builder\tool
s\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\lib
.py", line 301:
    force=force,
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\pac
kage.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\managers\pac
kage.py", line 483:
    vcs = VCSClientFactory.newClient(tmp_dir, url)
", line 61:
    self.check_client()
  File "c:\users\johna\.platformio\penv\lib\site-packages\platformio\vcsclient.py
", line 127:
    "Please install Git client from https://git-scm.com/downloads"
Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
fysetc_f6_13                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       IGNORED
STM32F1_base                  IGNORED
STM32F103RC_bigtree_512K_USB  IGNORED
STM32F103RE_base              IGNORED
STM32F103RE                   IGNORED
STM32F103RE_bigtree           IGNORED
STM32F103RE_bigtree_USB       IGNORED

STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_lite                IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
STM32_hal                     IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED
==================== 1 failed, 0 succeeded in 00:00:04.332 ====================

wieltar, 2019-12-02:
Have you restarted your PC/visual studio after installing git? It adds values to your path which only register to open programs when you've restarted your PC

Johnathan2581, 2019-12-02:
yes, So I have had some success the file now complies, but I have run into new problems with this firmware ... When I go to put the sd card into the board and turn on the printer it shows a blue led flashing for a couple of seconds along with the red light then I get a solid red light with no LCD except the blue backlight.
when I take out the sd card and check it on my laptop the firmware.bin file is there along with a firmware file in all caps. I Don't really know where to go from here. I'll attach my current working compile of the firmware.
[Marlin-2.0.x-SKR-E3-DIP.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3910097/Marlin-2.0.x-SKR-E3-DIP.zip)

closed 2019-12-04 by Johnathan2581