bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #42: Bigtreetech Marlin need update

ICGE6544, 2019-12-02:
Branch BIGTREETECH-SKR-E3-DIP-V1.0/Firmware/Marlin-2.0.x-SKR-E3-DIP/ need to be updated.

It would be nice from a member of Bigtreetech to update the marlin for the SKR mini E3 dip (the one for the skr mini e3 v1 is up to date).
If I compile the actual firmware for the SKR mini E3 dip with Atom and PIO 4.1 it causes a blue screen (brick) on my Ender 3, not the case with the one for mini E3 v1.It is not a problem caused by Platform io version, Marlin package should be updated.
So please update Marlin package for the mini SKR E3 dip, thank you.

TheZoker, 2019-12-05:
Update is done:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/commit/37e0af8e8e0d83ef535132595e23c3c7a5338a04

oldman4U, 2020-06-08:
Hello ICGE6544.

Could you please close this ticket.

Thank you

ICGE6544, 2020-06-09:
solved

closed 2020-06-09 by ICGE6544