bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #44: Blue screen (brick) on Ender 3 When upload new  Firmware

fabienroots, 2019-12-04:
When I compile the actual firmware for the SKR mini E3 dip with Vscope and PIO 4.1 it causes a blue screen (brick) on my Ender 3. When i compiled this 2 month ago : No problem.

I dont unerstand why, i can't build a new firmware .... i'm not the only one in this case.

Can you do something ?

Best regards

Johnathan2581, 2019-12-05:
I found that I had to get the latest ver of marlin then reconstruct it from their ie going through and setup up everything. I have tried using the Bigtreetech firmware but ended up getting your current error the only problem with using the lastest ver of marlin and redoing the config files is that the auto home doesn't work for some reason probably a setting I missed.

hopes this helps in some way. Here is my build for the SKR E3 Dip V1.0 feel free to edit it :) to get your printer working. 

[Marlin-bugfix-2.0.x.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/3924841/Marlin-bugfix-2.0.x.zip)

oldman4U, 2019-12-10:
Hi Jonathan2581.

I can confirm what you wrote. Using Marlin and doing the config from scratch works. Using the precompiled firmware also works, but using the E3 DIP firmware results in an unusable Ender-3.

@fabienroots: You will get a blue screen and a never ending beep, in case the stock CR-10 display is not activated in the firmware.

oldman4U, 2020-01-29:
Hello fabienroots.

Do we still need this ticket? If not, maybe you can close it.

Thank you

oldman4U, 2020-06-08:
@thinkyhead (Please close this ticket due to inactivity)

Thank you

thinkyhead, 2020-06-14:
Only @fabienroots or @bigtreetech can close this issue.