bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #45: Skr e3 dip Usb not detected.. driver not working

hopatz, 2019-12-06:
Hi
I try install maple driver from this github but only detected as unknown devices..
Please help

oldman4U, 2019-12-10:
Hi hopatz.

What would you like to do with the USB port?

 As far as i know you can use it with Printrun and print directly from Cura, but you will not be able to see the SD card like a drive of your computer and you will not be able to do firmware updates - again, based on the practical experience i have with this board.

hopatz, 2019-12-15:
My tmc2130 + skr e3 now working fine
But still got message tmc connection error, i just wanna to diagnostic why its happened

oldman4U, 2019-12-18:
Hi.

I get the TMC Connection Error as soon as i have the computer connected via USB and Cura v4.4 opened. Having Cura closed the time i start the printer solves the problem.

oldman4U, 2020-01-29:
Hello hopatz.

Is this issue solved in the meantime or do you need any further help?

boz3d, 2020-01-29:
Hi, I think you have an wrong mail address I have no issue with the boards! Brboris-------- Ursprüngliche Nachricht --------Von: oldman4U <notifications@github.com>Datum: Mi., 29. Jan. 2020, 10:22An: "bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0" <BIGTREETECH-SKR-E3-DIP-V1.0@noreply.github.com>Cc: Subscribed <subscribed@noreply.github.com>Betreff: Re: [bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0] Skr e3 dip Usb not detected.. driver not working (#45)Hello hopatz.
Is this issue solved in the meantime or do you need any further help?

oldman4U, 2020-01-29:
Hi boz3d.

Not sure why you got my answer, but it was for hopatz, the owned of this ticket.

Sorry.

oldman4U, 2020-06-08:
@thinkyhead (Please close this ticket due to inactivity)

Thank you

closed 2020-06-09 by hopatz