bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #52: USB Connection problem on MAC OS

shantionee, 2020-02-20:
Hello everybody! 
I have been using SKR E3 DIP (Marlin bugfix 2.0.x) on Ender 3 Pro for more than two months now. Now there is a need to connect via usb to my iMac (Mojave 10.14.6) to work in Repetier Host (1.2.0). 

But after spending about four hours, it did not begin to work. It seems that the connection with the board is established, the message in Repetier is "Connection opened", but the data or commands are not received and are not processed by the board. 

There are only two messages in the log: "Attempting to connect to printer", "Connection opened". All sent commands simply hang in idle and are not sent. I configured baud rate 115200 as indicated in the firmware. Port: usbmodem14302. I tried to install various drivers, but this did not fixed the issue. I used five different usb cablesto understand that this is not cause the problem. I tried to connect with the power supply turned off and on. No difference.

Please help, who knows what could be the problem.

shantionee, 2020-02-21:
After 6 hours of searching, I have some results on my problem. In search, already pretty exhausted, I decided to download all the old versions of Repetier Host for Mac. And sorting through and unsuccessfully installing one version after another, I found that the installed version of 2012 - 0.53 sees my printer, connects to it, sees the firmware and can fully control it without any problems! It was really unexpected. I knew that the problem was software, but that I was lucky to find a solution there, I did not expect. 

Of course, this does not solve my problem completely, because I can not use the latest version of the Repetier Host, but perhaps this gives certain hints for solving. 

This version uses the same settings as in future versions, the parameters are still the same. Which allows us to conclude that this is not the connection settings to blame.

shantionee, 2020-03-01:
Is there any official or suitable drivers for SKR E3 DIP on Mac OS? It seems that the problem is in the missing drivers. And the problem will not be solved by itself.

oldman4U, 2020-03-06:
Hi shantionee.

macOS provides the access and the drivers to the hardware on the mac and in contrast to other operating systems, there is no need to install any drivers. The applications then access the hardware through those parts of the macOS, so in case this does not work, the application itself is very likely the reason it does not work. I do not believe that BTT can change this, but Repetier will have to to it.

oldman4U, 2020-06-08:
Hello shantionee.

Could you please close this ticket.

Thank you

closed 2020-06-09 by shantionee