bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #53: SKR E3 dip Display bleibt Blau Board reagiert nicht

ericD-prog, 2020-03-03:
Hallo ich habe ein Problem..
Ich habe die Marlin kompiliert an meinem Ender 3 mit dem SKR E3 DIP mit TMC 2208 V3.0 im UART Modus. 
Jetzt nach einigen test Drucken habe ich das Problem, das wenn ich den Drucker an schalte, das Display Blau bleibt und die 3.3v LED als auch die Status LED dauerhaft leuchten. Was ist das für ein Problem? Die Verbindung per USB funktioniert nicht, auch nicht das resetten. Was kann ich nun machen?

oldman4U, 2020-03-06:
Hi Eric.

Ich befürchte du wirst in englisch schreiben müssen, wenn du eine Antwort auf deine Frage haben möchtest. Oder ich schaffe es dir zu helfen;-)

Versuch doch bitte einmal das USB Kabel zu entfernen und wenn das noch nicht hilft, die Firmware neu zu laden.

Hope this helps, pls let me know

oldman4U, 2020-04-15:
Hi ericD

Do we still need this ticket: Please close it in case the problem is solved in the meantime.

thx

chrisconnew, 2020-04-15:
Anyone else finding it impossible to unsubscribe from this github? No
matter what i try i still get emails coming through

oldman4U, 2020-04-15:
Have you tried the unsubscribe link in the reply?

chrisconnew, 2020-04-15:
Yeah. Plus the watching options on the website.

It'll be quiet then suddenly I'll start getting emails

closed 2020-07-26 by ericD-prog