bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #54: What firmware should I install

aISergey, 2020-03-04:
Tell me please. I bought an E3 DIP TMC2208 UART. When the printer starts up, it displays the message: "TMC Connection Error". What firmware should I install:
firmware-tmc2208-chip-on-bottom.bin
or
firmware-tmc2208-chip-on-top.bin

aISergey, 2020-03-07:
On the advice of the seller, I installed firmware-tmc2208-chip-on-bottom.bin and got a blue screen of death.

stockHuman, 2020-04-01:
After setting your jumpers to UART, flash [this file](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/blob/56bb4e56ac75b802b99a15611351fe062762531f/Firmware/firmware-tmc2208.bin) from pull #57.

oldman4U, 2020-06-08:
Hello aISergey.

Could you please let us know the status of this ticket.

Thank you

aISergey, 2020-06-09:
Hello oldman4U!
I'm sorry. The issue can be close, I made the firmware manually from the original Marlin.

oldman4U, 2020-06-09:
Hi.

Good to hear you could solve the problem!;-)

Unfortunately i can not close it, but only the user which opened it. So could you please close it.

Thank you

closed 2020-06-09 by aISergey