bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #55: Cannot Compile with A4988 on Z-Axis and TMC2208 on X and Y axes

goldengitboi, 2020-03-08:
I am trying to run my SKR E3 DIP with TMC2208 drivers on the X and Y axes, and A4988 drivers on my Z axis and Extruder. When I try to compile without making any other changes I get an error related to the serial.h file. I can't tell exactly what it is as that file isn't annotated.  
[image]
I've tried running TMC2208 on X,Y and Z axes, but the heatsink on the Z axis Driver runs into the cooling fan and won't let me bolt the lid down, and using a smaller heatsink causes the driver to malfunction. I've tried compiling in both VSC and Atom with tall the same dependencies. 
Any help is greatly appreciated.

oldman4U, 2020-06-08:
Hello goldengitboi.

Could you please let us know the status of the reported issue.

Thank you

goldengitboi, 2020-06-08:
I ended up frying the circuit board trying different commands. nothing worked so I started using an skr v1.3 board.

oldman4U, 2020-06-09:
Sounds changing the hardware solved the issue, great.

Could you please be so kind and close this ticket. Seems you do not need it anymore.

Thank you

goldengitboi, 2020-06-09:
This is not a solution. It's a capitulation. What I did won't help someone who has the same issue, and reading this will likely convince other people not to buy the SKR mini DIP.

oldman4U, 2020-06-09:
I use a SKR E3 DIP v1.0 with 2208s for all motors since almost a year now. Started with pre v2 Marlin and currently with Marlin 2.0.5.3 without any issue. So something else must be different which caused the issue you have seen. I also had bad experience using A4988 drivers together with an SKR Pro, but the 2208 based config works great.

Would you like to keep this ticket opened, so you can try to fix it, or would it be OK for you to close this ticket and use your SKR?

closed 2020-06-15 by goldengitboi