bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #56: Bad hotend temperature reading

3DSAT, 2020-03-10:
Hi:

I´m trying to solve this issue, 
ADC readings from hotend are not stable after some time printing good.
Temperature reading are wrong, you can see 210..209..199...205..209...211..etc
I have this issue with 2 diferent boards and 2 diferent printers, same issue.
All related hotend parts are new ( Hemera one, and Titan Aero), and the issue start after 2 or 3 weeks of good work.
PID is also not the problem
I tested Marlin 2.0.0. 2,0.1, and 2.0.4 with the same problem after 2 weeks.

Any ideas ??

oldman4U, 2020-06-08:
Hello 3DSAT.

What is the status of the reported issue?

3DSAT, 2020-06-08:
Hi, the 2 boards are working good after replace C16 capacitors for low ESR ones.
You can close the request if you want.
Thx an regards.

oldman4U, 2020-06-09:
Hi.

Thank you for the update. Unfortunately i can not close it, but only the person who started the ticket. There is the Close button right besides the Comment button.

Thank you

oldman4U, 2020-06-15:
@bigtreetech Could you please close this unneeded ticket. Thank you

3DSAT, 2020-06-15:
Closed 
Regards

closed 2020-06-15 by 3DSAT