bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #59: SD cart is not present in the menu

TOMDED, 2020-03-23:
I've just set up the configuration file to TMC 2208 with UART, uncomment manual mesh bed leveling and compiled. Every thing seems to be ok. Printer is moving all axes, homing ok, manual bed leveling ok. But can't find SD cart in the menu. Init Media is not initializing SD cart. Checked SD cart support in configuration files - also uncomment.
https://youtu.be/3FGq2QuHyu4
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4368910/Configuration.zip)

Viking117, 2020-04-04:
try 
#define SERIAL_PORT -1
#define SERIAL_PORT_2 2

BearBroendum, 2020-04-14:
@TOMDED . 
I´m having the same issue with my printer. Can not communicate with the board either on sdcard or mini-usb.
Have you come closer to a solution?

TOMDED, 2020-04-30:
There was a SD card problem. After change the SD card everything went to ok.

closed 2020-04-30 by TOMDED