bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #61: Extruder not working

moboch, 2020-04-11:
just received my Skr mini e3 dip today and went ahead and installed it. Then started a test print but at some point (ca. Half through) it just stopped extruding. This happens when printing from SD and via USB. I also tried different slicers and speeds but nothing helped. What is also really weird is, when enter G1 E100 F100 to the console, it retracts 10Omm rather than extruding it. I already increased the motor current a bit and tuned my Esteps. What could be causing this?

MohanMaker, 2020-04-15:
Hi, please attach the firmware files that you are using we can help troubleshoot. Or, try going into the firmware folder in this repository and flashing one of the precompiled firmware files to see if that helps your situation. Thanks!

moboch, 2020-04-15:
I already use the pre-compiled firmware from here.

petrce, 2020-04-19:
I confirm the same problem, I used the latest firmware-tmc2208.bin from this repo
I have SKR-E3-DIP-V1.1 with 4 tmc2208

petrce, 2020-04-19:
possible because of the bug in LIN_ADVANCE?
see: https://github.com/MarlinFirmware/Marlin/issues/17306

moboch, 2020-04-19:
Jup, it was linear Advance. I don't even know why this is enabled by default, since it is known to not properly work with tmc2208s, and these are the drivers they ship the board with.

closed 2020-04-19 by moboch

petrce, 2020-04-19:
please can you describe how you disabled it? is M900 K0 in code enough?

moboch, 2020-04-19:
I don't know if that's enough. I just re- flashed the firmware and disabled it in Configuration_Adv.h

MohanMaker, 2020-04-20:
Thank you for making us aware of this issue. I've submitted a pull request to this repository with the corrected "firmware-tmc2208.bin" file and have also added this as an issue to the troubleshooting section of the README.