bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #67: Cant find software for E3 DIP V1.1

Pretous, 2020-05-04:
Hello, i know this software work for V1.1 too, but you can add any notification about it. I spent whole day to find that software from V1.0 fit for V1.1

MohanMaker, 2020-05-13:
Pull request #70 fixes this.

txt4nk, 2020-05-14:
Pretous, I will compile a firmware for you if you tell me what your setup is (including all mods)

slippy3002, 2020-07-14:
TXT4NK would you compile a firmware for me?
Ender 3
E3 DIP V1.1
TFT35 v3.0
TMC2209 Drivers
all stock besides that

txt4nk, 2020-07-14:
Sure, ill compile it tomorrow morning when I get to work.

slippy3002, 2020-07-14:
> Sure, ill compile it tomorrow morning when I get to work.

Have you gotten the chance today?

txt4nk, 2020-07-15:
Im sorry, today turned out to be an absolute circus at work. I'll get it done tomorrow.

txt4nk, 2020-07-15:
This should do the trick. While you update your board firmware, I would also update the TFT's firmware if you haven't already. Below is a link to BTT's guide on updating firmware, and a link to a zip file with the project files and a pre-compiled firmware binary file for you.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

[Marlin-bugfix-2.0.x_slippy3002.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4925786/Marlin-bugfix-2.0.x_slippy3002.zip)

slippy3002, 2020-07-16:
txt4nk Thanks!  you are the best!

txt4nk, 2020-07-16:
No problem

slippy3002, 2020-07-16:
txt4nk The stepper motors are backwards and there is no jerk...

txt4nk, 2020-07-17:
I'll reverse the steppers tomorrow. Can you verify if the extruder is reversed please

slippy3002, 2020-07-17:
Yes the extruder is backwards too

txt4nk, 2020-07-18:
Give this one a shot.

[Marlin-bugfix-2.0.x_slippy3002.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4942144/Marlin-bugfix-2.0.x_slippy3002.zip)

slippy3002, 2020-07-22:
Sorry for the late reply.  That worked well.  Thatnk you again!  You should make this available for everyone.

txt4nk, 2020-07-29:
> Sorry for the late reply. That worked well. Thatnk you again! You should make this available for everyone.

I have added repo's on my github for my personal printers. I will start to do that for these builds that I do for you guys as well, when time permits of course. Hard to find time when I work 70+ hours a week.

oldman4U, 2020-07-30:
@Pretous 

The reported issue is solved. Please close this ticket.

Thank you

oldman4U, 2020-09-18:
@bigtreetech

Please close this ticket

closed 2020-10-10 by Pretous