bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #71: SKR E3 DIP v1.1 + bltouch doesn't work

palal2, 2020-05-13:
Hi All,

I was installing SKR E3 DIP v1.1 + bltouch 1.2 (connected to z stop + servo) on Ender 3
And iam using Marlin 2. 

I didn't able to make the Bltouch working .. When powering on the printer the bltouch is deploying twice and then the red light is on and the pin is up.

1. On homing or on bed leveling the bltouch pin remains up and the red led is on   
2. On the printer menu the bltouch commenteds do nothing.
3. I was enabling "BLTOUCH" and "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" on Marlin Config

Some questions: 
1. Dose "New version V1.1: Modified the SERVO port" effects the configuration ? 
2. Iam using BLTOUCH v1.2 - should i change something to the config using this ver ?
3. What iam doing wrong ??? 😢

I think its a config issue because the BLTOUCH deploys twice on power on. 

Thanks !

txt4nk, 2020-05-13:
I too am having this issue. I have two SKR E3 DIP V1.1 boards (one spare, both present this problem) on an Ender 5 Pro w/ a BLtouch 3.1. I have tried vanilla marlin 2 (official release & bugfix) and also the V1.0 E3 Dip firmware posted here on github.

The BLtouch does its power up self test just fine (deploy, stow x2) but when homing or commanding the probe to deploy, stow or self test there is zero action.

I too am uncommenting #define BLtouch & #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN.

What gives, BTT?

txt4nk, 2020-05-13:
I have this semi figured out! The SERVOS pins were moved around on V1.1.
I just pulled out my multimeter and tested continuity between the servos pins and the ground input from the PSU. The ground pin on the V1.1 is now the far right (right next to the TFT and EXP1 connections). I am working on figuring out the +5V and Signals right now. Will post up results shortly.

[image]

palal2, 2020-05-13:
[image]
According to the pin layout the gnd is on the right side .
I connected the bltuoch in this way

txt4nk, 2020-05-13:
then you need to swap the other two wires. Mine is up and running. Probe works as it should.

txt4nk, 2020-05-13:
Another thing I noticed, the BTT website states the V1.1 comes with the STM32F103RC processor, mine have STM32F103RE's. May be work re-defining that in your platform.io file if you haven't already.

palal2, 2020-05-13:
Wich two wires ? The 5v and control?
What is the right order?

palal2, 2020-05-13:
> Another thing I noticed, the BTT website states the V1.1 comes with the STM32F103RC processor, mine have STM32F103RE's. May be work re-defining that in your platform.io file if you haven't already.

How did you check Processor type ?

txt4nk, 2020-05-13:
From left to right Yellow\Orange (signal), Red (+5V), Blue\Brown (Ground).

txt4nk, 2020-05-13:
It's printed on the processor chip.

[image]

txt4nk, 2020-05-13:
Mine is working beautifully now. I hope your check on the wiring and changing the processor help you as well.

palal2, 2020-05-13:
> then you need to swap the other two wires. Mine is up and running. Probe works as it should.

The probe is connected like you said, Control , 5v , gnd 
and still it deploy twice on power on and dose not deploy on homing 

still having issue, any more ides ?

txt4nk, 2020-05-13:
Can you verify your bltouch version? I don't recall ever seeing a 1.2..

palal2, 2020-05-13:
[image]

> Can you verify your bltouch version? I don't recall ever seeing a 1.2..

The wired thing that is deploying on power on ...

txt4nk, 2020-05-13:
Since I have no experience with that version BLtouch, I'm afraid I don't know what I can offer as far as help here. I would recommend purchasing a V3.1 and swapping it in with the current wiring.

bsl-ender, 2020-05-17:
Hello.
I will join this topic because there is no point in opening another one.
I have a problem with the correct compilation of Marlin on DIP v1.1 with stepdrive 2209 and bllouch v3.1.
Is anyone so good with Configuration.h and Configuration_adv.h or ready firmware.bin?

txt4nk, 2020-05-17:
I can compile one for you. Post you printer specs (model, all mods, bltouch offset measurements, etc)

bsl-ender, 2020-05-17:
> I can compile one for you. Post you printer specs (model, all mods, bltouch offset measurements, etc). Also. Look at the processor on your skr board and let me know if it is a RET6 or RCT6

Ender 3 235x235x250, bltouch 3.1, NOZZLE_TO_PROBE_OFFSET { -40, -5, -1.85 }, skr e3 DIP stick 2209, RET6

txt4nk, 2020-05-17:
Alright, just got to work. I'll get it compiled and posted shortly.

txt4nk, 2020-05-17:
here ya go bsl-ender. I uploaded the firmware.bin in a zip file because it wouldn't let me upload the binary file. I also uploaded the entire project folder with all of the configurations set in case you want to tweak any settings and try to compile it yourself. 

[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640569/firmware.zip)
[Marlin-bugfix-2.0.x_bsl-ender.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640570/Marlin-bugfix-2.0.x_bsl-ender.zip)

bsl-ender, 2020-05-17:
> here ya go bsl-ender. I uploaded the firmware.bin in a zip file because it wouldn't let me upload the binary file. I also uploaded the entire project folder with all of the configurations set in case you want to tweak any settings and try to compile it yourself.
> 
cut

Do not work correct ;( 

https://www.youtube.com/watch?v=vJmZMjZ_F1E

txt4nk, 2020-05-17:
Did you cut the DIAG pin on the 2209? You will have homing issues until you cut the DIAG pin, its a very well documented issue with the 2209's & the SKR E3 DIP (and any other SKR boards with replaceable drivers) 

https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf

bsl-ender, 2020-05-17:
> Did you cut the DIAG pin on the 2209? You will have homing issues until you cut the DIAG pin, its a very well documented issue with the 2209's & the SKR E3 DIP (and any other SKR boards with [...]

yes

txt4nk, 2020-05-17:
alright, I'll reverse the stepper motor and we can see if that works.. No idea why we would need to do that but it did move the wrong direction.

txt4nk, 2020-05-17:
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640647/firmware.zip)
[Marlin-bugfix-2.0.x_bsl-ender.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640648/Marlin-bugfix-2.0.x_bsl-ender.zip)

bsl-ender, 2020-05-17:
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640647/firmware.zip)
> [Marlin-bugfix-2.0.x_bsl-ender.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640648/Marlin-bugfix-2.0.x_bsl-ender.zip)

better :D but not perfect :P 
I connect BLT 3pin to servo 2 pin to Zstop ... that is correct? 
And small issue ... Z is reverse goint up not down :P

txt4nk, 2020-05-17:
Correct, the 3 pin connects from inside of the board to the outside in this order: Yellow\Orange (signal), Red (+5V), Blue\Brown (Ground). Yellow\Orange should be on the inside, blue\brown should be closer to the outside, next to the TFT\Screen ports.

Here are updated firmware and project files to correct the Z movement. Once you get things working, you need to get the hot end heated up and see if the extruder is reversed or not. If it is, let me know and I will compile the firmware again with a fix for that as well.

[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640686/firmware.zip)
[Marlin-bugfix-2.0.x_bsl-ender.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640687/Marlin-bugfix-2.0.x_bsl-ender.zip)

bsl-ender, 2020-05-17:
Extruder is reverse :P

txt4nk, 2020-05-17:
haha okay standby.

bsl-ender, 2020-05-17:
> haha okay standby.

OK ... If you live in Poland or German I buy you beer ( if your age is correct to drink alkohol )

txt4nk, 2020-05-17:
My friend, if I was still in Germany I would most definitely take you up on that offer! I'd love a nice cold Warsteiner Dunkel right now!

Here are your revised files =)
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640741/firmware.zip)
[Marlin-bugfix-2.0.x_bsl-ender.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4640742/Marlin-bugfix-2.0.x_bsl-ender.zip)

bsl-ender, 2020-05-17:
> My friend, if I was still in Germany I would most definitely take you up on that offer! I'd love a nice cold Warsteiner Dunkel right now!

I don't drink alcohol. I don't like it; P If you will be in Germany, please contact: D

txt4nk, 2020-05-17:
For sure!

Let me know how that setup works for you.

bsl-ender, 2020-05-17:
Work good :D 
CHEP Cube is OK :D

txt4nk, 2020-05-17:
> Work good :D
> CHEP Cube is OK :D

Excellent!

Steve2017, 2020-05-22:
I also have the DIP 1.1 and a BLTouch - v3.1 bought from Creality.  I have the latest Marlin installed and while the BLTouch will work to run levelling, it will not work as an endstop for Z-homing. I have tried about 15 different firmware setting combinations, including one from someone who says their board and BLT work fine.

On the old Creality (silent) board, the BLTouch worked both as an endstop for Z-home and for levelling. On the DIP I have had to reinstall the microswitch for Z-Home.  Which way is it working on yours?

Steve2017, 2020-05-22:
> For sure!
> 
> Let me know how that setup works for you.

I also have the DIP 1.1 and a BLTouch - v3.1 bought from Creality. I have the latest Marlin installed and while the BLTouch will work to run levelling, it will not work as an endstop for Z-homing. I have tried about 15 different firmware setting combinations, including one from someone who says their board and BLT work fine.

On the old Creality (silent) board, the BLTouch worked both as an endstop for Z-home and for levelling. On the DIP I have had to reinstall the microswitch for Z-Home. Which way is it working on yours?

txt4nk, 2020-05-22:
Do you have the 2 pin plug with the white and black wires plugged into the probe port or the z axis end stop port?

Steve2017, 2020-05-22:
> Do you have the 2 pin plug with the white and black wires plugged into the probe port or the z axis end stop port?

Plugged into the Probe port. What happens is if there is nothing in the z-endstop socket, the machine will not lower the hotend when you try to home Z - it just sits there.

If I plug the BLTouch into the z-endstop socket (& enable that in Marlin - or not) it will attempt to home, but will not respond to the probe being triggered. It would hit the bed if I allowed it.

If I plug both in their respective sockets it will home to the microswitch endstop and level on the probe.

I assume from your question that yours will Z-Home with the BLTouch.

Steve2017, 2020-05-23:
> Another thing I noticed, the BTT website states the V1.1 comes with the STM32F103RC processor, mine have STM32F103RE's. May be work re-defining that in your platform.io file if you haven't already.

I also noticed your comment here - Mine also has the RE chip but the ENV is set at RC - I notice there is reference further down in PlatformIO to the right chip - does that make a difference?

txt4nk, 2020-05-23:
> Plugged into the Probe port. What happens is if there is nothing in the z-endstop socket, the machine will not lower the hotend when you try to home Z - it just sits there.
>
> If I plug the BLTouch [...]

Ah, yes you need to have the bltouch plugged into the z end stop, not the probe. In the firmware you need to uncomment use z end stop as probe. That will solve the problem with the nozzle colliding with the bed. And yes, the wrong processor could make a huge difference. If you'd like, tomorrow when I get to work I can also compile a firmware for you if you leave a comment with your exact setup.

Steve2017, 2020-05-23:
I have tried it with the BLT plugged into the Z-endstop - and have uncommitted the line you mention - it will not recognise the BLT probe triggering - but if you have a better set-up - fantastic

I still have the stock hotend & bed, with the Creality BLT bracket for offsets. (I think it’s -41, -10, -2.5) I have the TFT E3, so the CR10 setting works for the display. Everything else is pretty stock or is configured in Cura.

If you do have the time - thank you for the .bin - but could I also ask for a copy of your configuration.h, config_adv.h and the platformio file you are using - I’d like to see how your settings vary from mine. (Using Latest Marlin release BTW - not bugfix)

txt4nk, 2020-05-23:
> I have tried it with the BLT plugged into the Z-endstop - and have uncommented the line you mention - it will not recognise the BLT probe triggering - but if you have a better set-up - fantastic
> 
> I still have the stock hotend & bed, with the Creality BLT bracket for offsets. (I think it’s -41, -10, -2.5) I have the TFT E3, so the CR10 setting works for the display. Everything else is pretty stock or is configured in Cura.
> 
> If you do have the time - thank you for the .bin - but could I also ask for a copy of your configuration.h, config_adv.h and the platformio file you are using - I’d like to see how your settings vary from mine. (Using Latest Marlin release BTW - not bugfix)

I'll definitely get one compiled for you and upload the entire project folder. I hope it works for you. Are you positive that you've got the wiring correct for the 3 pin connector? When you go to the SKR boards, you have to swap the yellow and red wires around in the connector so the red is in the center position. The 3 pin connects from inside of the board to the outside in this order: Yellow\Orange (signal), Red (+5V), Blue\Brown (Ground). Yellow\Orange should be on the inside, blue\brown should be closer to the outside, next to the TFT\Screen ports.

Steve2017, 2020-05-23:
Wiring has been checked against the 1.1 PDF many times - and is wired as you state. The fact that it tests and actually levels makes me think the wiring is okay. I gather the two-pin probe connection is also polarity sensitive and that seems okay.

I just tried reflashing with the probe uses zmin_endstop defined, and the plug in that socket - same result as before.... On Z-home the pin deploys but the board ignores when triggered.

I then replaced the BLT into the probe socket and the micro switch into the z-endstop socket (without changing the firmware). On z-home the pin will deploy but it is ignored in favour of the micro switch.

Interestingly I see Chris Riley has recently uploaded a YouTube video where he talks about still using the micro switch, so maybe........

Thanks for your time.

txt4nk, 2020-05-23:
Have you tried using the bugfix version? Also I'm going to head into the office here soon (took the morning go catch up on sleep). I want to verify your setup before I compile the firmware.

Ender 3
Stock hot end and hot bed
Creality bltouch mount
TFT e3 screen

What drivers are you running on your E3 DIP 1.1?

Steve2017, 2020-05-24:
I have 2208 drivers, but about to change to 2209.

Just hold off - I have changed the environment to the RE version, and have redone all the cable connections -  and late last night when I tried, I got a response from the BLT on z-homing.  However I had two different results on two different runs and figured I was too tired and needed to do this with fresh eyes to make sure why this has happened.

I can’t believe that the environment change would make this difference because they both point to a lot of the same files/sub-routines.... but I couldn’t chase all the paths down (or understand them)

txt4nk, 2020-05-24:
Hey progress is great. I can definitely see the environment change altering the functions. If there was no difference between them, there probably wouldn't be an environment call out for each of them.

Steve2017, 2020-05-24:
I have done a couple of runs this morning and it both homed and levelled on the BLT, but my offsets are wrong.  I want to compare the two Config files with the old ones line for line to see what I have changed beside “Probe uses z minstop”.

The compare function in Word is useful for this.

txt4nk, 2020-05-24:
A quick way to get a rough z offset that I typically do with great success is set it to -10, auto home, manually move the z axis via the menu until a piece of paper makes contact with the nozzle and the bed, but not so much the theres a ton of drag. Whatever your z height is, deduct that from the -10 and you've got a damn close z offset that you can fine tune by printing a zig zag pattern.

Steve2017, 2020-05-24:
It’s the x & y offsets that are off at the moment- it’s going off the bed on levelling.  I just need some time free to redo those & reflash.

txt4nk, 2020-05-24:
Is the probe in the center of the bed when it auto homes too, or only when its probing?

Steve2017, 2020-05-24:
Centre for Z-Home.  I have Safe_homing defined which I believe forces it to the centre of the bed.

My birthday party duties are over so I will have another shot at this. I’ll fix up the offsets, compare the old and new config files and then I will also do a build using the old RC environment to see if that was the cause of the BLT fault.

txt4nk, 2020-05-24:
Let me know your findings, I'd like to know if the environment is the culprit as well.

Steve2017, 2020-05-25:
The change of environment did not make a difference - BLT is still working properly... plugged into the Z-Endstop.  I have not run a print job yet, but can't see why it wouldn't work.

I have gone back through a lot of my firmware changes and I can't find the triggering issue - except I did get some intermittent results on a couple - for example the offsets.  They didn't seem to take and the BLT was trying to measure off the plate - I rebuilt without a Config change and it was happy.  I have no idea why.

This has me scratching my head.  The only thing I can think is that I had a bad connection or I was hallucinating when I thought I changed the wires around on the probe plug. (But I'm sure I did) .

I will save the HERO bin file and will try a couple of other version on things.

Thanks for your help - I think you have earned a therapy degree here.

txt4nk, 2020-05-25:
haha I like it, the hero bin file.

Try getting some prints out on it, I am currently having issues with my printer while using the BLtouch.. when its not active, I get great first layers but when it is, I get some pretty wild patterns.. I thought the bed may have been warped so I replaced the bed and the exact same pattern is in the exact same spot, and I can disable ABL, and the prints are beautiful..

1st bed: 
[image]
2nd bed:
[image]

Steve2017, 2020-05-26:
That is frustrating.  I am now going through the re-configuring with the new board, so I will do all my calibrations (e-steps, flow-rates etc) again. There's a couple of hours there... so I will probably have the same issues are you.

As a thought with my variable results, I thought I will initialise the EEPROM to clear out any settings, put them in the firmware and go from there. I can always change later.

BTW  "HERO bin file" is taken from video production where the original version of the "golden take" is the one you see as the hero of the production. Doesn't mean as much these days with digital, but on tape, you needed to protect the original copy. A .bin file that works is gold.

txt4nk, 2020-05-26:
> That is frustrating. I am now going through the re-configuring with the new board, so I will do all my calibrations (e-steps, flow-rates etc) again. There's a couple of hours there... so I will [...]

An init is always step 1 after flashing new firmware. In fact my steps are init and then a power cycle just to make sure there's no garbage in there.

I can totally agree, a working binary file is stupid hard to achieve right now lol

txt4nk, 2020-05-29:
Went UBL this morning and did about 20 minutes of manual mesh leveling in the spots I was having issues and I have a near perfect 200x200 first layer. With this, I can go back into my mesh and manually change the few problem area's on the 200x200 print to achieve the perfect first layer. Thanks for nothing, bilinear lol

Blizzard1984, 2021-05-24:
> My friend, if I was still in Germany I would most definitely take you up on that offer! I'd love a nice cold Warsteiner Dunkel right now!
>
> Here are your revised files =) [...]

Hello. I have a board e3 dip v1.1 (ender 3pro tmc2208, direct drive bmg v6) and I tried your firmware. It works but can't be changed from probe offset. I don't know what to do. Could you please help me? I can't compile firmware. I can upload bin to sd card. Would you do firmware for me. Bin? I would be very grateful. Sorry for the translation. I used a google translator. I am from the Czech Republic. Have a nice day. Martin

donjakobo, 2021-10-30:
> haha I like it, the hero bin file.
>
> Try getting some prints out on it, I am currently having issues with my printer while using the BLtouch.. when its not active, I get great first layers but when [...]

This might be a late reply -- It's not that a strange pattern, your Z-offset is too close, and the imperfections of your bed (there will always be some) cause squishing and this shows up as this ripple pattern in 1st layers and later as almost an over-extrusion. Adjust your z-offset for the prob more (give more space) and rerun a small 35x35 block to see