bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #72: When printer is heating the control board\screen switches between menus

txt4nk, 2020-05-13:
Ender 5 Pro
SKR E3 DIP V1.1 w\ TMC2208 UART's

When my printer's heaters are activated, the screen acts like I am pressing on the knob, it continuously goes through menu's. Makes it impossible to set things during a print.. Any ideas? My SKR E3 Mini did it as well.. But my ezboard lite did not.

oldman4U, 2020-05-14:
Hi.

Seems there are other users having similar issues, where the PCB of the screen touches the metal frame of the display enclosure.

Hope this helps

txt4nk, 2020-05-14:
I will see if I can put some sort of spacers between the board and frame. Thanks so much for your response!

txt4nk, 2020-05-15:
I cannot believe how simple this fix is... lol it works! Screen has not been in contact with the frame all day and it has not spazzed out once. I am in the process of designing and printing a buffer\stand off for the screen bracket right now so I can reconnect it to the frame with non conductive material so the possibility of EMI traveling through the frame is gone.

@oldman4U Thanks again for the resonse, it is greatly appreciated.

closed 2020-05-15 by txt4nk