bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #73: BTT SKR E3 DIP -V1.1 no longer reacts to anything

Alessiano, 2020-05-22:
I got a weird problem.
No matter what I do, the board stops responding.
It started by reloading the bootloader as described in error [#2](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/2).
I was able to successfully flash the original bootloader and firmware with a STLink v2.
but then I still couldn't install a new firmware via the SD card.
I tried again to flash only the bootloader.
since then nothing works anymore.
The USB can't detect anything, not even an "unknown device".
With a STLink v2 I cannot connect to the board anymore.
The display always shows "no connection to printer".

I have no idea what else I could do...
Does anybody have any more ideas?

oldman4U, 2020-06-08:
Ciao Alessiano.

This sounds like a hardware issue. Have you been able to solve the problem in the meantime?

Please let us know

Thank you

Alessiano, 2020-06-09:
Unfortunately, I have not been able to correct the error so far.

I have the assumption that the basic information, that tells the chip what kind of chip it is, are no longer present.

To get the information back on the chip, you have to "unbrick" it.

But, That exceeds my knowledge.

oldman4U, 2020-06-09:
But the board has been working "before"? I mean it should work out of the box and has been a 1:1 replacement for my Ender-3 Pro.

Alessiano, 2020-06-10:
yes, it worked
i could also update the firmware via the sd card
at least the first time
then i could not update again, because the bootloader disappeared as described in [#2](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/2)

oldman4U, 2020-06-10:
I would try to get a replacement. A bootloader does not disappear without a reason.

Alessiano, 2020-06-13:
I have contacted support.
I am curious to hear what they have to say.

oldman4U, 2020-06-13:
I would recommend to contact the dealer and not the manufacturer.

You need a replacement.

Alessiano, 2020-06-13:
Yes, I have contacted the dealer support

oldman4U, 2020-06-15:
Hi.

Any news?

Alessiano, 2020-06-15:
Hi.
No answer so far...

oldman4U, 2020-06-15:
;-(

Alessiano, 2020-06-17:
I have received the following answer:
(I've checked all that in the past)

Hello,
USB does not display the motherboard, you can reinstall the driver, this is the driver link:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/tree/master/Firmware/Marlin-2.0.5_SKR-E3-DIP/Marlin/src/HAL/LPC1768/win_usb_driver
Unable to connect to the printer, you can try the following steps:
1. Make sure that serial port 1 is enabled in the marlin firmware and the baud rate is set to 115200
2. Please confirm that the connection between the motherboard and the touch screen is
  Motherboard touch screen
   5V<->5V
   GND<->GND
   TX<->RX
   RX<->TX
   RST<->RST
3. Make sure the baud rate of the touch screen is also 115200
Unable to update the motherboard via SD card. This is the SD card update procedure:
Try to format the SD card, choose FAT32 as the file system,
Then copy the firmware.bin file to the SD card of the motherboard,
Put the SD card back into the SD card slot of the motherboard,
Then power on the motherboard, press the reset button on the motherboard, and wait 15 seconds.
The motherboard can successfully update the firmware.

Yours sincerely,
BIQU customer service

I hope they will replace the motherboard

oldman4U, 2020-06-17:
Yes. These are the standard recommendations...

So you told them that you tried this all?

Alessiano, 2020-06-18:
Yes.
And this is the second answer:

Hello,
Still unable to update the firmware? Does the indicator flash when updating the firmware?
If it still shows that the printer cannot be connected, you can use a multimeter to check whether the RX and TX pins of the TFT terminal are connected to GND. If it is on, it is a short circuit problem.
You can send me the test results as pictures or videos, and I will carry out after-sale forensics for you.
You can replace a TFT touch line without short circuit.
After changing the touch line, it still cannot display normally. The chip may be abnormally hot. We will carry out after-sales for you.

Yours sincerely,
BIQU customer service

I'll try that out right now.

oldman4U, 2020-06-21:
Hi.

Any news?

Please let us know

Thank you

Alessiano, 2020-06-24:
I am currently creating photos for the support team of the message on the display, the cabling and the firmware.

I'll report as soon as there is any news.

oldman4U, 2020-06-26:
Hehe, a picture says more than 1000 words. At the moment i am waiting for a spare encoder for my TFT35 E3 which have had issues from the beginning and now does not jump back after pressing the knob...;-(

oldman4U, 2020-06-29:
Maybe it is too late now, but have you tried to change the fuse?

oldman4U, 2020-07-11:
Hi. Still no news!!??

Please let me know

Alessiano, 2020-07-14:
After being asked to measure the TFT pins for the fifth time, I gave up...
I bought a new motherboard.

closed 2020-07-14 by Alessiano