bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #74: failed to build firmware

drmshady, 2020-05-25:
i tried to compile marlin file and i got this error
{
	"resource": "/c:/Marlin-2.0.5_SKR-E3-DIP/.vscode/c_cpp_properties.json",
	"owner": "c:\\Marlin-2.0.5_SKR-E3-DIP",
	"severity": 4,
	"message": "Cannot find \"C:\\Marlin-2.0.5_SKR-E3-DIP\\include\".",
	"startLineNumber": 9,
	"startColumn": 17,
	"endLineNumber": 9,
	"endColumn": 53
}

txt4nk, 2020-05-25:
This is just a wild shot off the hip.. But having stuff on the root of the C: always ends up having problems with authorization\admin rights. Move the folder to your desktop and try again.

H-DieterK, 2020-05-25:
It is filed on the desktop too, and thanks for this tip!

txt4nk, 2020-05-25:
It was worth a shot!

drmshady, 2020-05-25:
it is failed from desktop but with different error message
{
	"resource": "/c:/Users/mshad/Desktop/Marlin-2.0.5_SKR-E3-DIP/.pio/libdeps/STM32F103RC_btt/Adafruit BusIO_ID6214/Adafruit_I2CDevice.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'",
	"startLineNumber": 160,
	"startColumn": 79,
	"endLineNumber": 160,
	"endColumn": 79
}
{
	"resource": "/c:/Users/mshad/Desktop/Marlin-2.0.5_SKR-E3-DIP/.pio/libdeps/STM32F103RC_btt/Adafruit BusIO_ID6214/Adafruit_SPIDevice.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'SPI_BITORDER_LSBFIRST' was not declared in this scope",
	"startLineNumber": 115,
	"startColumn": 23,
	"endLineNumber": 115,
	"endColumn": 23
}
{
	"resource": "/c:/Users/mshad/Desktop/Marlin-2.0.5_SKR-E3-DIP/.pio/libdeps/STM32F103RC_btt/Adafruit BusIO_ID6214/Adafruit_SPIDevice.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'SPI_BITORDER_MSBFIRST' was not declared in this scope",
	"startLineNumber": 37,
	"startColumn": 43,
	"endLineNumber": 37,
	"endColumn": 43
}

txt4nk, 2020-05-25:
try using this environment in your patformio.ini file.

STM32F103RC_btt_512K

txt4nk, 2020-05-25:
if that doesn't work, post your detailed setup (printer, all mods, dip board version, tmc driver type, bltouch or not, if so include your offsets) and I'll see if I can get one compiled for you

drmshady, 2020-05-25:
thank you
iam using ender 3 with BIGTREETECH-SKR-E3-DIP-V1.1 motherboard
2208 tmc driver 
i am using bl touch z offset -3.5
i am planning to using filament run-out sensor connect to e0 stop

txt4nk, 2020-05-25:
> thank you
> iam using ender 3 with BIGTREETECH-SKR-E3-DIP-V1.1 motherboard
> 2208 tmc driver
> i am using bl touch z offset -3.5
> i am planning to using filament run-out sensor connect to e0 stop

I need all three offsets (or at least the X and Y) for the bltouch.
Lets get the printer running with your current setup before you try to add the run out sensor.

txt4nk, 2020-05-25:
Looks like you are not the only one. I am having the issue too. Hopped over to the marlinfirmware github page and sure enough, it is a known bug that they are working on right now.

drmshady, 2020-05-25:
X=-44, Y=-16

drmshady, 2020-05-25:
can you send me the file before compiling so i will try to see whats the problem with mine

txt4nk, 2020-05-25:
It's nothing wrong with what you're inputting. programs like atom (platform.io) periodically reach out to the internet to download the latest bootloader (adafruit) libraries and the latest version of bug fix (which the btt files are based off of, they were updated recently) are calling out to a library that is not compatible with these boards. looks to be a programming error from what I read, which is why this older version of bug fix I have is working, it is using the older library (before the programming error).

txt4nk, 2020-05-25:
inside the following zip is the entire project file so you can load it into your compiling program, as well as a compiled firmware binary file.

[Marlin-bugfix-2.0.x_drmshady.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4679062/Marlin-bugfix-2.0.x_drmshady.zip)

drmshady, 2020-05-25:
thank you for your help

txt4nk, 2020-05-25:
My pleasure! This stuff isn't just something you know over night, I know it can be super frustrating to run into issues like this so wherever I can help, I do.

drmshady, 2020-05-25:
it worked 
ifoolowed the post you mention and i replaced Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
to
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip
thank you

txt4nk, 2020-05-25:
awesome, I'm glad that helped!

closed 2020-06-12 by drmshady