bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #75: Stepper Motors getting Very hot

Damin69, 2020-05-26:
I just got the E3 Dip combo with the 2209 drivers. My issue is that my Stepper motors are getting very hot hot enough you can not touch them. I had the same issue with the Standard E3 Mini V1.2 on this this Unit. I have 2 Ender 3 pros. One is running all stock stepper motors and I have no issues at all with it and it is running an E3 Mini v1.2. The unit I am having problems with stepper motors getting hot I have a Hermera Direct Drive on it. The Motor on the Direct drive does not get hot only the X, Y and Z stepper motors. I am running Firmware 2.0.5 that I downloaded April 25th when I installed the board in this machine.

oldman4U, 2020-05-27:
Hello Damin69.

What are the current settings you are using in config. adv. for the motors and what is their rated current?

Damin69, 2020-05-27:
I am using the stock Settings for Ender 3 on both of my machines. Current on x, y and Z is 580 current on the Extruder is 650 and is being set through Firmware 
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y580 Z580
Recv: echo:  M906 T0 E650

oldman4U, 2020-05-27:
Thats fine, but the question is, what the rated A of the motors is. If this is less than 1A you will have to go down, even the settings are from the stock Ender 3. I do not believe that there is another way to get them cooler, except for active cooling of course.

Damin69, 2020-05-27:
I have 2 Ender 3s with SKR boards This one overheats. I do not understand why? I run the same firmware settings on both. I run the same Gcode on both of them this one will overheat.

oldman4U, 2020-05-28:
Have you every tried to swap the motors to see if the problem follows? If you have 2 printers, this would be the easiest way to check.

oldman4U, 2020-05-28:
And i assume, you still use the motors which came with the Ender 3's on both machines!?

Damin69, 2020-05-28:
Ok I could see this if it was one motor but NOT all of them. And only does it with the SKR board. NOT the stock boards which I have moved around.

Damin69, 2020-05-28:
Lets ask this if the Drivers are in UART and you are setting the current in Firmware do you still set the VM on the driver chip?

oldman4U, 2020-05-28:
Not sure why you do not want to make the easiest test of all, but it is your decision. We also do not know if the motors are ALL the stock motors or different motors.

Using 2209 the current is defined in Configuration_adv.
Using 2209_Standalone the current is set in hardware.

Damin69, 2020-05-28:
Easy test? Well I think I found the problem on face book group. I am using Octoprint Via USB someone mentioned covering up the 5volt pin on usb and try. And low and behold NO hot motors. Hmm.

oldman4U, 2020-05-29:
This sounds like the solution, great. Maybe you could close this ticket and make a new one, so that other users also know about this problem. This would be great.

Thank you

Damin69, 2020-05-29:
**Ok here is what I have found. It was not a current setting in my case. I am using an SKR E3 DIP Not sure if this would be same on the E3 mini but I am going to assume so. I was using a Pi4 with octoprint to control my Ender. After a post on the Facebook group someone suggested putting tape over the 5Volt postive Pin on the USB cable coming from the PI. I did that and now all my stepper motors are all runing nice and cool. 

This may just be a Pi 4 issue with the 5 volt pushing to much current as my other set up with SKR board does not have this issue but that is a Pi 3B+. So time will tell good luck hope this helps someone.**

closed 2020-05-29 by Damin69