bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #77: CE3, E3 DIP V1.1, Extruder works for purge line, but does not work for print

txt4nk, 2020-06-12:
As stated in the title, I have an Ender 3 w/ an E3 DIP V1.1 (TMC2208's in UART). Just installed this into one of the printers in my farm, which had a creality non-silent main board in it before and was printing just fine. After the install, everything works great except for the fact that when I start a print the extruder will extrude just fine for the purge line, but stops extruding after that. What could cause this? I am using the exact same file that is being used in another Ender 3 right next to it, that has an E3 Mini in it.

oldman4U, 2020-06-12:
Hello txt4nk.

Which display are you using, where do you print from (on board, TFT, SD, USB?) and could you please upload the gcode file which fails on your machine.

Thank you

txt4nk, 2020-06-12:
Printing directly off the SD card. Here's the gcode.
[6 post face shield_ender 3.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4773217/6.post.face.shield_ender.3.zip)

oldman4U, 2020-06-12:
SD card of the TFT or the E3 DIP mainboard?

txt4nk, 2020-06-12:
the E3 DIP's, sorry I didn't specify before.

oldman4U, 2020-06-12:
no problem. Your gcode has a M204 S1000 where it stops. Have you tried to replace it for a test:

Legacy M204 S<accel> is deprecated. Use separate paremeters M204 P<accel> T<accel> instead.

txt4nk, 2020-06-12:
I have not, however that is something that my slicer is injecting into the gcode. these legacy m204 are also injected into the gcode of other machines running an E3 Mini, E3 DIP and a  ezboard lite.. none of them are having this issue. I will remove it and try a print, but I'm not very hopeful it will be the fix as the others are working with it just fine.

txt4nk, 2020-06-12:
removing that line worked.. I wonder why its happening only on THIS dip board.. I also do not see any way to stop prusaslicer from injecting that in sadly.

txt4nk, 2020-06-12:
Actually I just figured out how. There are acceleration settings in the printer section, and there are also "advanced" acceleration settings in the print section. Removing values from the print section remove that line between the purge and the print. Still very odd that that only one printer is effected by it. 

Thank you oldman4u

closed 2020-06-12 by txt4nk

oldman4U, 2020-06-13:
I guess this is related to the Merlin FW running on the hardware. Using
deprecated functions is never a good idea.

Glad i could help. Please close the ticket in case you haven’t done so.

Thank you and happy printing