bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #78: Example configuration for v1.1 RE6 board with TMC2130 drivers?

ckuethe, 2020-06-13:
Can you post the exact configuration files used to compile `firmware-TMC2130.bin`?

ckuethe, 2020-06-15:
@MohanMaker ping? How did you compile the binaries you uploaded?

MohanMaker, 2020-06-17:
Hello! I simply used the configuration files in "Marlin-2.0.5_SKR-E3-DIP" in the repository and modified (and uncommented) these lines to say TMC2130 as shown.
#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE TMC2130
#define E0_DRIVER_TYPE TMC2130

But I believe there must be something else that you need to change in addition, since bigtreetech recently posted a commit fixing the "sw spi" for the TMC2130 pre-compiled files. I think that means software SPI, but I am not sure exactly what is is or how to configure it (I don't use TMC2130 or sensorless homing) so maybe someone more knowledgeable can give some input.

oldman4U, 2020-09-29:
Please help the community and close the ticket once you do not need it anymore. Thank you

ckuethe, 2020-10-01:
The issue isn't fixed yet. The exact configuration - including the previously mentioned sw spi fix - used has not been posted.

oldman4U, 2020-10-02:
So you are waiting since 4 months for something which will come if someone from the community will do it!?

oldman4U, 2020-10-11:
@ckuethe 

The problem is that you do not understand how this works here. This is open source software and also pre-compiled examples are made by a user. So you finally ask another user to do what you could do by yourself.

Please help the community and close the ticket once you do not need it anymore. Thank you