bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #79: New skr-e3 / extruder Problems

Crazywelder72, 2020-06-13:
I just installed a brand new board and the extruder now isn’t functioning properly. All it does is wiggles about 1/32” forward and backward. No idea why this would happen. The firmware is bugfix2-0.x. Someone suggested I try to switch the middle wires on the connector as if it’s out of phase but it was working properly on the original board.

oldman4U, 2020-06-14:
Hi Crazywelder72.

We would need more information about your setup, like what kind of divers you are using and also pictures which show your jumper setup as well as information about the motors you are using could be helpful. Please attach also the platform.ini, the configuration.h and configuration_adv.h file here, so we can check them.

Thank you

Crazywelder72, 2020-06-14:
Thanks Oldman4u, 

So I will try my best to fill in some blanks but I cannot answer everything you asked as idk. So the machine is a Ender 3 and I bought it on amazon supposed to be plug and play. It was loaded with the most recent version of marlin. Jumpers i didn’t do anything and other than the board and the 4 heatsinks there wasn’t anything but the duck in the box and the sd card. Another fb member sent me his bin file to reflash the firmware with however it did nothing. I tried moving the extruder connector to another encoder position and use the control to move it however same issue. Lastly I replaced the original board to see if somehow I cooked the extruder motor but using the original board everything worked fine. I’m returning this board on amazon and getting another but if it happens again 🤷🏻‍♂️

oldman4U, 2020-06-15:
Hi.

Even the hardware is a direct replacement for the Ender-3, the software and the remaining hardware like stepper motors have to be configured as well as installed properly to make it work.

You will have to read the manual and to watch some YouTube videos to be able to replace the mainboard, or you find someone to do it for you.

To be able to help you here, we would at least need to know which driver motors you installed the heatsinks onto. And you would have to remove/install some jumpers on the mainboard.

Please let me know and we can help

Thank you

Crazywelder72, 2020-06-15:
Hi again,
There is no manual that came with the board and I’ve watched so many YouTube videos I can’t fathom what is what. People start talking about audrinos and hex files all of that I don’t have the capability to do. However I was originally pointed to getting this board as is has a boot loader already installed for simple reflashing upgrades. The heat sinks are installed on the chips as shown on the boards website. The heatsinks are not touching any adjacent components on the board.

Attached is a pic of the board (after I removed it) 
And as you can see there’s no jumpers on it because...well.... none came in the box and nothing said to the contrary they would be required . 
[image]

oldman4U, 2020-06-15:
This picture says more than 1000 words - for sure;-) This all can be very confusing, I totally agree. But it is like always. What is much too much for one person is the starting point for the next person.

You posted your issue at the wrong place. You own a BTT SKR MINI E3 v1.2 which has the drivers mounted fix on board, and the place you started your ticket is for the DIP version of "the same board", which has changeable motor drivers but does not have other features your board has.

Both boards are a nice replacement for the original mainboard of the Ender-3, so there is mouthing wrong with the board you bought. The only problem we have now is, that i do not own the board you bought, but the DIP version, so i can not really help, sorry.

The user manual is the pdf you can find here on GitHub (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2manual.pdf) and this is the correct place to ask questions related to your board: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues

I am sorry that i can not help you, but please copy and paste your initial text together with the picture into a new ticket at the link from above and i hope someone there can help you with your problem. If NOT, come back here and i will try to help, OK!?

Kind regards

oldman4U, 2020-06-17:
Hi.

Have you started a ticket in the correct Repository in the meantime and could someone help?

Crazywelder72, 2020-06-17:
Thanks again for your guidance.. I clicked on the link to the other thread however it specifically said to use Facebook group for those boards 🤷🏻‍♂️. I’m already in the Facebook group and wasn’t getting much help at all except from one person. I am still awaiting the arrival of the new board tomorrow so I can check what he mentioned.. actually measure the mA at the motor and see if it matches what the output says on the display (I think 900ma for the extruder if I recall). I didn’t even know I can modify that output via the display panels if required. Thank you for following up again..

Crazywelder72, 2020-06-21:
> Hi.
> 
> Have you started a ticket in the correct Repository in the meantime and could someone help?

As it turns out as suspected the board was junk from the beginning. I installed the second board last night and the extruder is functional.

oldman4U, 2020-06-21:
DOA.;-( This is the worst case which happen to a new user. You always believe you make something wrong and in reality the hardware is bad...

I am happy to hear that the new board is working fine. Could you please help us to maintain this system and close the tickets which you do not need anymore.

Thank you and

HAPPY PRINTING

Crazywelder72, 2020-06-21:
Thanks again... 👍

closed 2020-06-21 by Crazywelder72