bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #81: Marlin 2.0.5 doesn't compiling for SKRs boards

VijitCoder, 2020-06-14:
I have tried a lot of configurations from the original Marlin repo.

I ended in this repo. 

Downloaded and tried to compile "as is" - FAIL. Tried to change in platformio.ini `default_envs`:
- STM32F103RC_btt (that is default from this repo)
- STM32F103RC_btt_512K
- STM32F103RE_btt

The compilation is **always fails**. Different errors, but the same result - no firmware.

At the same time I see in this repo prepared firmwares, which is obviously were compiled with the Marlin sources from this repo. So, how did you manage to do this?

ckuethe, 2020-06-15:
Maybe a duplicate of #74 which references MarlinFirmware/Marlin#18081 ? I think you just have to change one line in your `platformio.ini `file

VijitCoder, 2020-06-15:
Thanks a lot!

I spent a few hours, fighting till deep of the night, but didn't find the answer. And what is turned out.. A wrong dependency.

Thanks.

---

Just for the protocol. File `platformio.ini`, change Adafruit_MAX31865 link from the "master" to 1.1.0:
```
[common]
lib_deps =
...
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip
```

closed 2020-06-15 by VijitCoder