bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #83: No USB Serial

Adiwan, 2020-06-25:
Hello.
I tried to connect the printer via USB in order to make a PID autotune, however upon connecting the printer via USB (printer was turned on before) the screen stops updating the image, the PC cannot establish a serial connection, and the extruder stepper motor makes an alarm noise.
The problem occurs with the current firmware with the TMC2208 stepper drivers compiled in.

oldman4U, 2020-06-25:
Hi Adiwan.

Do you have Cura opened on the same computer? If so, close it and try it again.

Please let us know

Adiwan, 2020-06-25:
I have used Pronterface and bCNC and the arduino IDE serial monitor in order to establish a connection.

oldman4U, 2020-06-26:
All apps at the same time or one after the other?

So you finally have been able to connect?

Adiwan, 2020-06-26:
Every program after another. No more than one program was running at the same time that could connect to the serial interface.
And no. It didn't work.

oldman4U, 2020-06-26:
OK.

Which display do you have and is it illuminated when you connect the USB cable even when the printer is turned off?

Which message do you get, when you click on Connect in Pronterface?

Adiwan, 2020-06-26:
I have the default Ender 3 LCD display.
When the printer is off and then I connect the printer with the PC via USB, then 
the display turns on and shows only the Marlin boot screen for ever.
Pronterface freezes up during that time with no message.
Only when I pull the cable out pronterface will unfreeze and write an exception 
message that the device has been removed during establishing a serial link.

On Friday, June 26, 2020 3:07:36 PM CEST oldman4U wrote:
> OK.
> 
> Which display do you have and is it illuminated when you connect the USB
> cable even when the printer is turned off?
> 
> Which message do you get, when you click on Connect in Pronterface?

oldman4U, 2020-06-26:
OK, this means that the connection between the computer and the mainboard works in general.

I asked about Cura, because the issue you describe now (frozen UI) is exactly what happened when you had Cura opened, where a driver conflict caused this issue. I GUESS that something on your PC causes a similar issue, by grabbing the USB port in a certain way. Maybe you have another computer you can try?

oldman4U, 2020-06-29:
Hi Adiwan.

Any news regarding the reported issue?

Please let us know

Adiwan, 2020-06-29:
The problem persists with other PCs. I tried several Windows 10 and Linux devices. But they suffer all from that same issue.

oldman4U, 2020-06-29:
Do you compile the firmware by yourself and if so, which serial ports have you enabled?

Adiwan, 2020-06-29:
As I said it didn't work with the compiled one here on offer.
My own compiled firmware had the serial port not changed. Later I tested to other values but it didn't work either.
Effectively I did only change the thermistor type, max temperature, and added the PID autotune menu, as well as the stepper driver type.

oldman4U, 2020-06-29:
You have to have serial -1 activated in your firmware, otherwise the USB connector will not work. Make sure this is the case.

There is nothing else i can help with, so i hope someone else has an idea

oldman4U, 2020-07-03:
Hi. Any news regarding the reported issue?

Adiwan, 2020-07-04:
There is still the problem. Today I received a replacement board and I will test it very soon.

oldman4U, 2020-07-11:
Hi. Any luck with the replacement?

oldman4U, 2020-08-03:
@Adiwan 

Please let me know if you have been able to solve the problem, or just close this ticket.

Thank you

oldman4U, 2020-08-26:
Still not tested?

Please close this ticket in case the problem is solved

closed 2020-08-26 by Adiwan

oldman4U, 2020-08-26:
👍🏻