bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #84: DIP v1.1 MKS 2209 UART Connection error

MyqlGodfrey, 2020-06-27:
Vanilla compilation of marlin 2.x TMC Connection Error.
Vanilla compilation of marlin 2.x with TMCStepper pointed at BIGTREETECH.. TMC Connection Error.
Download of firmware2209.bin.. TMC Connection Error...

I have MKS 2209 steppers (I also bought some fysetc 2209s...but they appear to be total crap).. same issue with both.
Board + TFT3.5 e3 v3.0 connected with drivers through USB.. no PSU connected... not sure if that is the issue.
I am on assignment in Vietnam and cant get the standard SKR mini E3 (or I would)...so Now I have a board and 8 drivers that dont work...and thinking of getting 4 more..just to see if the 3D gods will smile on me.

oldman4U, 2020-06-27:
Hi.

Maybe you can attach your configuration files here.

What says M122?

MyqlGodfrey, 2020-06-27:
Attached are configuration and configuration_adv
[Configuration_files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4841415/Configuration_files.zip)

m122 shows all low and none of the drives has an address.. thank you so much for getting back to me.

oldman4U, 2020-06-28:
Hi.

From what I can see, software looks good, so you issue is likely hardware related.

I have no experience with the MKS drivers but use BTT 2208s, so if it helps i can check if i get the same TMC Connection error in case i disconnect the PS.

Why do you have no PS attached?

MyqlGodfrey, 2020-06-29:
I bought a new DIP board and TFT3.5 E3 so I have them on the bench and I am setting them up (in a comfortable environment) :)
I am about to give up and just use the existing melzi board (1.1.4) and Kipper.

oldman4U, 2020-06-29:
Good news.

When i power off my perfectly working E3 DIP/2208 and connect the USB, so the display is on, i get the same TMC errors and informations when executing an M122 like you get. So you have to connect a PS to get the drivers work, which makes absolutely sense keeping in mind the low voltage you get over USB.

So please connect a PS and try again. It is worth.

MyqlGodfrey, 2020-06-30:
You sir are a scholar and a gentleman.
I will try this weekend (I broke down and am using Klipper with Melzi now).. I will 'upgrade' this weekend and get everything stitched back up (still printing some screen holders for new TFT etc).
I really appreciate you pulling the power for me!
Have a great day,
M

oldman4U, 2020-06-30:
😊you are very welcome.

oldman4U, 2020-07-10:
Hi. Any news?

Please let me know 

Thank you

MyqlGodfrey, 2020-07-11:
Yay!  I had a bit of a scare (I started with the FYSETC v3.0 silent 2209s.. they are crap).  I switched to MKS 2209s and BAM it all started working.  Sorry for taking so long to get back, I had the opportunity to do a quick vacation (Hoi An) so I took it. 
Thank you so much for the follow up (and for pulling your power to test for me), much appreciated.
I just got a bunch of PETG I am excited to chew through.
Have a great day!
[image]

closed 2020-07-11 by MyqlGodfrey