bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #85: Visual Studio Code + PlatformIO Compiler Probleme

Tec-Mar, 2020-06-28:
Hallo,

Ich kann die Beispiel Firmware welche man hier herunterladen kann nicht Compilern.. es kommen immer folgende Fehler.

Compiling .pio\build\STM32F103RC_btt\src\src\HAL\shared\eeprom_spi.cpp.o
Compiling .pio\build\STM32F103RC_btt\src\src\HAL\shared\persistent_store_api.cpp.o
Compiling .pio\build\STM32F103RC_btt\src\src\HAL\shared\servo.cpp.o
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\onboard_sd.cpp:15:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\STM32F103RC_btt\src\src\MarlinCore.cpp.o
*** [.pio\build\STM32F103RC_btt\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt\src\src\HAL\STM32F1\onboard_sd.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\HAL_spi_L6470.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt\src\src\HAL\shared\HAL_spi_L6470.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\eeprom_spi.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from c:\users\hanis\desktop\marlin-skr-e3-dip\marlin\src\hal\stm32f1\hal.h:36,
                 from c:\users\hanis\desktop\marlin-skr-e3-dip\marlin\src\hal\hal.h:26,
                 from Marlin\src\HAL\shared\backtrace\../../../core/../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\../../../core/serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:29:
c:\users\hanis\desktop\marlin-skr-e3-dip\marlin\src\hal\stm32f1\fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt\src\src\HAL\shared\eeprom_spi.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt\src\src\HAL\shared\backtrace\backtrace.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\eeprom_i2c.cpp:31:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt\src\src\HAL\shared\eeprom_i2c.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\backtrace\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\unwmemaccess.cpp:18:
Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\servo.cpp:54:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt\src\src\HAL\shared\servo.cpp.o] Error 1
In file included from Marlin\src\inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\inc/../HAL/HAL.h:26,
                 from Marlin\src\inc/MarlinConfig.h:30,
                 from Marlin\src\MarlinCore.h:24,
                 from Marlin\src\MarlinCore.cpp:31:
Marlin\src\inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt\src\src\MarlinCore.cpp.o] Error 1
=============================================================================================================================================== [FAILED] Took 24.30 seconds ===============================================================================================================================================

Environment               Status    Duration
------------------------  --------  ------------
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt                **FAILED**    00:00:24.298
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      IGNORED
STM32F103RC_btt_512K_USB  IGNORED

========================================================================================================================================== 1 failed, 0 succeeded in 00:00:24.298 ========================================================================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

oldman4U, 2020-06-28:
Hi.

Bitte lade Marlin von hier herunter: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x. Bevor du kompilierst, stelle sicher, dass du (wie auf dem Screenshot zu sehen ist) alle PlatformIO Updates durchgeführt hast.

Hoffe das hilft

[image]

Tec-Mar, 2020-06-28:
Hy, habe jetzt mal mühselig die komplette Marlin Bugfix mit der Firmware Marlin verglichen und dementsprechend die Werte abgeändert! Nun klappt es endlich und das SKR Board kann super angepasst werden.. woran es nun genau lag gute Frage.. werde aber deine Version auch nochmal testen ob es vlt. an der älteren Marlin 2.0.5. lag. ODER aber was ich auch gefunden habe.

https://reprap.org/forum/read.php?415,874307 

Danke schonmal für die Hilfe!

oldman4U, 2020-06-29:
Es kommt öfters vor, dass alte Marlinversionen und neue Treiber für diverse Teile nicht zusammenpassen. Manchmal funktioniert es nicht einmal, wenn neu und neu verwendet werden;-( Dann hilft meistens Google weiter;-)

Bitte schliesse dieses Ticket wenn du es nicht mehr brauchst, danke.

closed 2020-06-29 by Tec-Mar