bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #86: BLTouch problem.

pevire, 2020-06-30:
Hello, I've compiled marlin from: BIGTREETECH-SKR-E3-DIP-V1.0/Firmware/Marlin-2.0.5_SKR-E3-DIP/ to use BLTouch.

X and Y homing works fine, but if I do a Z homing, the Y carriage goes to the right (even beyond of the limits making a "crack crack crack" sound) and I get this message:

"READ: Error:!! STOP called because of BLTouch error - restart with M999
READ: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)"

BlTouch is well connected and it works well with M280. If I compile Marlin without BLtouch, Z homing works well.

Any idea? Thanks.

pevire, 2020-07-01:
I edit. It is not a problem with BLTouch. Marlin in

oldman4U, 2020-07-01:
Hi.

I would attach the Marlin files so someone can check them.

pevire, 2020-07-01:
A problem with drive type. Sorry.

closed 2020-07-01 by pevire