bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #87: Its not compiling - gives error message on row "Adafruit_I2CDevice.cpp:160"

degustator, 2020-07-06:
Indexing .pio\build\STM32F103RC_btt\lib294\libUSBComposite.a
.pio\libdeps\STM32F103RC_btt\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio\libdeps\STM32F103RC_btt\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: **error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'**
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);

degustator, 2020-07-06:
Ok found a fix for this somwhere else:

Just needed to change:

From:
;Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip"

To:
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip

oldman4U, 2020-07-07:
Hi. Have you tried to use the bugfix version of Marlin?

oldman4U, 2020-07-10:
Hi.

I compiled without the error above using the Bugfix version today. Could you please close this ticket.

Thank you

degustator, 2020-07-10:
I'm closing ticket, but it could be good to update this repo, instead of going somewhere else

closed 2020-07-10 by degustator