bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #88: Compile error with fresh everything

bkrowiak, 2020-07-14:
Coincidentally new install of win10 and everything else is giving me this when I try to compile 2.0.5 for my E3 DIP
All I'm attempting to do is change settings in configuration.h to account for some new hardware. Previous firmware was probably 2.0.0
[image]

oldman4U, 2020-07-15:
Hi.

Please try this Marlin version: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

and change the environment to "BIGTREE_SKR_PRO in platformio.ini

It it fails the first time, try it a second time

Please let me know

bkrowiak, 2020-07-16:
It got me through, but I had to change the environment and tool around with library dependencies. Thanks

closed 2020-07-16 by bkrowiak