bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #89: fatal error: libmaple/gpio.h: No such file or directory

mchaos, 2020-07-19:
I keep getting the following. Have the skr e3 dip v1.1, using Atom with platform.io on linux.

Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory

and 

*** [.pio/build/STM32F103RC_btt/src/src/HAL/STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp.o] Error 1

First I compiled the firmware as it was with no editing and it compiled fine. Then I went on to set up for TMC2130. I set all the settings to make this work.

Then I got the error.

So I went back. Removed the project to start from scratch. Re extracted the firmware in a new directory and tried to compile it as it was with no change...

Now I get the error all the time.

I downloaded Bugfix from Marlin github, moved the config files, still getting the errors.

Not sure how to get this working, I have been hammering at it since yesterday...

oldman4U, 2020-07-19:
Maybe you can attach your files here-

mchaos, 2020-07-19:
It the same files you have on the git hub. Unedited.... Nothing changed.. Will not compile.. SO... waste of time me uploading them when they are already here.

mchaos, 2020-07-19:
I downloaded straight from Marlin Github. Manually changed parameters in the config files as opposed to copying them. It compiled. However I made no other changes but setting the environment and board. I will let you know if setting it up completely with all needed settings works. Something is up with your config files I think.

mchaos, 2020-07-19:
I made all the edits to the firmware manually. I did not copy the config files over. It did compile and is working on board. Have not tested board, but it boots and I can connect with Pronterface. The downloaded FW from your git does not compile no matter what I do.  Only direct from marlin with everything manually changed.

oldman4U, 2020-07-21:
Hi.

If you look at the date the firmware in this repository has been made, you get an idea what the problem might be.

But I am happy you could solve the problem.

Please do not forget to close this ticket as soon as the reported issue is solved

Thank you

mchaos, 2020-07-25:
Still not solved. TMC2130 connection errors. I had to copy the Platform IO because other dependencies to make it work. When I use the Platform io with the options it needs, it will not compile. I get Marlin/src/HAL/DUE/fastio/../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory

oldman4U, 2020-07-25:
You wrote it works when you use uncompiled version from Marlin!???

mchaos, 2020-07-25:
it compiled, but now the TMC does not work. When I add the rest of the variables, it fails.. I am closing this one. we will talk about both in the other.

closed 2020-07-25 by mchaos