bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #91: My prints are coming out wobbly looking

shadowind98, 2020-07-24:
Ok so I installed the board correctly to my ender 3 pro, I then proceeded to download the recompiled firmware to my tf card, ( it’s the tmc2208 drivers) I renamed it to firmware.bin and then I put it into the printer it worked !!! I was so happy and so I test to see if the z axis was moving  and it did (my z axis driver blew from the stock board leading me to change it to the skr e3 dip v1.1 board), but then I started printing one of my prints and the print was leaning  not like from gravity but like as if it couldn’t keep it in the same position when printing.

shadowind98, 2020-07-24:
[image]

shadowind98, 2020-07-24:
[image]

shadowind98, 2020-07-24:
[image]

oldman4U, 2020-07-26:
Hi. I use the same mainboard and driver and can not see this problem. Have you ever tried to compile your own firmware using VSC and PlatformIO? Have you checked the hardware? Maybe swap the drivers of two axis.

shadowind98, 2020-07-31:
> Hi. I use the same mainboard and driver and can not see this problem. Have you ever tried to compile your own firmware using VSC and PlatformIO? Have you checked the hardware? Maybe swap the drivers of two axis.

so no I didn't compile my own firmware do to me not familiar with that process and I don't know what I should and shouldn't change. and when you mean by swapping the drivers you mean the drivers that can be taken off and on the board right ? so if thats the case then no I haven't done that, what would that do or confirm and thank you for responding to me  I really appreciate the willingness to do so

oldman4U, 2020-08-03:
Hi.

Sorry for the late answer....

Not compiling the own firmware means missing most of the fun;-)

Anyway, but you uploaded new firmware successful to the mainboard - which display are you using?

Swapping the drivers from one axis to another, for example x and y as well as z and E0. Doing so you could check if the problem changes or if it remains the same.

shadowind98, 2020-08-06:
sorry for late response but I'm using the default display for the ender 3 pro

oldman4U, 2020-08-06:
Ok. Have you tried to swap the driver or even the motors?

Maybe the driver blew for a reason which is still there.

oldman4U, 2020-08-13:
Hello @shadowind98 

Could you please answer my question?

Thank you

shadowind98, 2020-08-20:
sorry once again at the time I was in the middle of preparing for my sisters graduation so I was so bogged down with stuff
anyways so the driver didn't blow because when I installed the new board for the first time it made the prints wobbly over and over again, not to mention the whole reason I got the board was so that if the drivers blew I could replace them instead of the board, the board that was in the printer originally is the one that the z axis driver blew and so I bought a new stepper motor and and the new board so everything is new and supposed to work but I don't know

oldman4U, 2020-08-21:
Ok. Not sure what the reason could be. Have you uploaded the latest Marlin bugfix version to the mainboard?

shadowind98, 2020-08-24:
I see ok thanks

oldman4U, 2020-08-24:
Not sure which display you use but maybe you want to start the print from Pronterface to see if this makes a difference.

shadowind98, 2020-08-28:
its the regular display when you buy the ender 3 pro

shadowind98, 2020-08-28:
I have a Mac

oldman4U, 2020-08-28:
Then use Printrun. It is the same.

shadowind98, 2020-09-01:
ok ill check it out

AlexanderAmelkin, 2020-09-03:
Looks a lot like missed steps or some major mechanical problem.

Check that:
1. Belts are tense
2. Pulleys are firmly sitting on the motor shafts
3. Steppers current matches that of your motors. AFAIK Creality 42-34 steppers that are used for axes in Ender 3 and 3 Pro are rated 1.5A. Make sure you don't exceed that value, yet aren't too low. The default settings for TMC2208 in BTT's clone of Marlin are just 580mA. I would try setting them to 1000 and see if it helps. Can't tell for the precompiled firmware, but in stock Marlin those values are adjustable from the LCD menu.

shadowind98, 2020-09-16:
No didn’t get too work on it yet

shadowind98, 2020-09-16:
> Looks a lot like missed steps or some major mechanical problem.
>
> Check that:
>
> 1. Belts are tense
> 2. Pulleys are firmly sitting on the motor shafts
> 3. Steppers current matches that of your motors. [...]

Can you explain so I understand I’m still new to the terms and at this point I may try to have someone else try to fix it cause this is a bit advanced for me

federico77, 2020-09-28:
Hello, I am not sure if this is the same for you or if this offers some ideas about the problem but I noticed a similar behavior with my card.

Thing is: for me it only shows up if I turn it off, leave the USB cable connected to my Octopi print server (over RPI3) and then I turn the printer on again. In this case my prints come out wobbling (in the best case). If I completely turn off the printer by ALSO removing the USB cable to the print server, I see no issues at all.

Hope it helps.

oldman4U, 2020-09-28:
Good point but the user does not compile firmware. So using Octoprint is even a bit more complex. But I would pick up your idea and ask if something else is connected. E3 DIP does not offer many options but who knows...🥴

federico77, 2020-09-29:
To be honest I am not saying this was/is Octopi related: it feels more like something connected with the USB port powering the board while main power is missing. My idea was that having the 5v from USB but not the 12/24v from main was messing with the current running Marlin registers/memory... but I am no expert so please take this more as a feeling than a theory.

oldman4U, 2020-10-13:
Ok. So what is your plan?

shadowind98, 2020-10-13:
well at this point ill probably sell it and get the money to go towards a new better printer cause besides the help you guys have given me I also tried a 3d printer service company in Staten Island and they stoped doing 3d printer repairs and I even reached out to folks on craigslist ,no good so that's where I am now

closed 2020-10-15 by shadowind98