bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #92: Testing X connection... Error: All LOW

mchaos, 2020-07-25:
Trying to set up TMC2130.

Compiled successful and run M122...

		X	Y	Z	E
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	1436	1436	1436	1436
MAX current	2025	2025	2025	2025
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual	0/31	0/31	0/31	0/31
PWM scale	0	0	0	0
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
tstep		0	0	0	0
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false
off time	0	0	0	0
blank time	16	16	16	16
hysteresis
 -end		-3	-3	-3	-3
 -start		1	1	1	1
Stallguard thrs	8	8	0	0
uStep count	0	0	0	0
DRVSTATUS	X	Y	Z	E
sg_result	0	0	0	0
stallguard
fsactive
stst		*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

oldman4U, 2020-07-25:
Please close your unneeded tickets before open the next one. 

Thank you

mchaos, 2020-07-25:
both are valid issues

oldman4U, 2020-07-25:
How have you powered the mainboard?

mchaos, 2020-07-25:
12v.. I also flashed the firmware on your git and I get the same error.

oldman4U, 2020-07-25:
For me this looks like the drivers do not get enough power but i use 2208
and 24V so i do not have experience with this combination.

Can you attach the config files.

mchaos, 2020-07-25:
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/files/4976174/Configuration.zip)

mchaos, 2020-07-25:
The problem I see is that there is more in the platformio.ini from this firmware. Stock marlin FW is this:
[env:STM32F103RC_btt]
platform          = ${common_stm32f1.platform}
extends           = env:STM32F103RC
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
build_flags       = ${common_stm32f1.build_flags}
  -DDEBUG_LEVEL=0 -DSS_TIMER=4 -DHAVE_SW_SERIAL
monitor_speed     = 115200

And this is the entry in the one from this git.:
platform          = ststm32
board             = genericSTM32F103RC
platform_packages = tool-stm32duino
build_flags       = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DHAVE_SW_SERIAL -DSS_TIMER=4
build_unflags     = -std=gnu++11
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
src_filter        = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_deps          = ${common.lib_deps}
  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
lib_ignore        = Adafruit NeoPixel, SPI
monitor_speed     = 115200

But when I use the one from this git, it will not compile. I feel as though some driver or dependency is not downloading.

oldman4U, 2020-07-25:
Hi.

I used the bugfix version from GitHub Marlin and your configuration files and only changed the platform.ini (STM32F103RC_btt) and it compiled successful on my system.

mchaos, 2020-07-26:
It compiles just fine... BUT TMC2130 will cont connect.  It obviously compiles if I am able to do an M122 command through pronterface and give you the output. I need to get it to connect to TMC2130. That's the problem.

mchaos, 2020-07-26:
Use this in platformio.ini and it will not.

oldman4U, 2020-07-26:
But the driver registers and connection status looks like they do not get enough power. Disconnect the 12v and use USB power only and i believe you will get the same m122.  As an advice i would also start with 16 microsteps and hybrid treshold off. My setup with 24v and btt tmc2208 works great.

oldman4U, 2020-07-26:
What will not? Like i wrote the bugfix version works in poi with you config
files.

mchaos, 2020-07-26:
So I am using the latest bugfix. Drivers are getting plenty of juice. I think I figured it out. When I bought these drivers, I did not know that there were jumpers on the boards. I bought these boards back when they were still fairly new, and had not tried to implement them. The original TMC2130's did not have jumpers. These are not original and the retailer I got them from provided no info on this. Looking at the same board now has documentation that was not available at the time. Mine came setup for stand alone mode.

Having the old version up is confusing. I think you should just have the latest drivers and latest config file examples because the firmware you have up will not even compile due to its age and the progressment of marlin.

I still do not know if they will work, but I can say the firmware is compiling as expected. It attempts to connect to the TMC2130's  through SPI but the drivers are being jerks right now due to them being in stand alone mode.

I compiled for standard A4988, configured board jumpers to step mode and popped in some A4988 drivers.  I did confirm that the board is working well.

Thanks for trying to help. I will use your micro step advise. I'll reconfigure the stepdrivers, board and firmware tonight and give it a try. I really hope these things work. Let you know.

mchaos, 2020-07-26:
So I get an error all high.. but it's a different result.. Damn hard to solder those jumpers.. I don't have a microscope. I did my best but maybe I didn't get a full joint. I only tested one. This is not a firmware issue at this point.

closed 2020-07-26 by mchaos

oldman4U, 2020-07-26:
Maybe it would be better to use it in stand alone mode? But stealth chop is
it worth. The loudest part of my printer is the 20.- euros Noctua fan😁

I hope you can activate them!

mchaos, 2020-07-27:
I tried stand alone. It just buzzed loud. didn't move.

mchaos, 2020-07-27:
maybe these drivers are garbage...

oldman4U, 2020-07-27:
There is another user having problems with the same driver but another
board right now.

Re: [bigtreetech/BIGTREETECH-SKR-PRO-V1.1]  SKR PROv1.2 FYSETC TMC2209v3.0
by UART (#157)

oldman4U, 2020-07-27:
Look. Another user has issues with the same driver.

Re: [bigtreetech/BIGTREETECH-SKR-PRO-V1.1]  SKR PROv1.2 FYSETC TMC2209v3.0
by UART (#157

Yes you are right. ;-(

You set the jumper on the board right?

oldman4U, 2020-07-27:
It is ANOTHER MAINBOARD but describes you driver:

https://3dwork.io/en/complete-guide-skr-v1-4-and-tmc2209

mchaos, 2020-07-27:
I have everything set up the way it is outlined.